sriharshaarangi/BatteryChargeLimit #4: doesn't work for Wireless Charging

bam80, 2017-04-22:
Nexus 4,
the program works fine if charging through micro USB.
On [Wireless Charger](https://en.wikipedia.org/wiki/Qi_(standard)), the charging starts regardless of battery level.
I could assist to debug it, if necessary.
Thanks

milux, 2017-05-30:
I need some more information, especially filtered logcat showing the debug output of the app when the device starts to charge.
But to be honest, it looks like a flawed implementation of the wireless charging with your device. If we don't find any other reliable way to detect the charging state, we might not be able to fix this.
Have a look at the end of this post for logcat instructions: https://forum.xda-developers.com/showpost.php?p=72218650&postcount=581

milux, 2017-07-28:
I will close this due to inactivity. If the issue shows up again and you can provide sufficient information, please notify me and I will re-open it.

bam80, 2017-07-28:
OK, thanks. Sorry for ignoring your request, will send the logs later

bam80, 2018-01-21:
For some reason, wile on wireless charger, the device seems continue charging independently of Control file value:
```
mako:/ # cat /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                     
1
mako:/ # cat /sys/class/power_supply/battery/status                                                                                                                                                             
Charging
mako:/ # echo 1 > /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                
mako:/ # cat /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                     
1
mako:/ # cat /sys/class/power_supply/battery/status                                                                                                                                                             
Charging

mako:/ # cat /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                     
1
mako:/ # echo 0 > /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                
mako:/ # cat /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                     
0
mako:/ # cat /sys/class/power_supply/battery/status                                                                                                                                                             
Charging
mako:/ # echo 1 > /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                
mako:/ # cat /sys/module/pm8921_charger/parameters/disabled                                                                                                                                                     
1
mako:/ # cat /sys/class/power_supply/battery/status                                                                                                                                                             
Charging
```

Here is the log:
```
bam@darkstar:~$ adb logcat -e "(CHAR|POW)"|tee logcat.txt 
01-21 02:57:35.914  3551  3551 I R.System: Received intent android.intent.action.ACTION_POWER_CONNECTED
01-21 02:57:35.916  3551  3551 I S.StatsService: ACTION_POWER_CONNECTED
01-21 02:57:36.297 27975 27975 D Power State: ACTION_POWER_CONNECTED
01-21 02:57:36.818 27975 27975 D Charging State: CHARGE_FULL 165872925
01-21 03:14:02.232 27975 27975 D Charging State: CHARGE_STOP 165872925
01-21 03:14:12.174 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 165872925 (Delay: 1000)
01-21 03:14:15.227 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 165872925 (Delay: 2000)
01-21 03:14:22.170 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 165872925 (Delay: 4000)
01-21 03:14:42.178 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 165872925 (Delay: 8000)
01-21 03:15:12.184 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 165872925 (Delay: 16000)
01-21 03:16:01.777  3551  3551 I R.System: Received intent android.intent.action.ACTION_POWER_DISCONNECTED
01-21 03:16:01.778  3551  3551 I S.StatsService: ACTION_POWER_DISCONNECTED
01-21 03:16:01.850 27975 27975 D Power State: ACTION_POWER_DISCONNECTED
01-21 03:16:02.151  3551  3551 I R.System: Received intent android.intent.action.ACTION_POWER_CONNECTED
01-21 03:16:02.157  3551  3551 I S.StatsService: ACTION_POWER_CONNECTED
01-21 03:16:02.194 27975 27975 D Power State: ACTION_POWER_CONNECTED
01-21 03:16:12.196 27975 27975 D Charging State: CHARGE_STOP 43511999
01-21 03:16:15.306 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 43511999 (Delay: 1000)
01-21 03:16:22.174 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 43511999 (Delay: 2000)
01-21 03:16:30.244  3551  3551 I R.System: Received intent android.intent.action.ACTION_POWER_DISCONNECTED
01-21 03:16:30.256  3551  3551 I S.StatsService: ACTION_POWER_DISCONNECTED
01-21 03:16:30.276 27975 27975 D Power State: ACTION_POWER_DISCONNECTED
01-21 03:17:18.265  3551  3551 I R.System: Received intent android.intent.action.ACTION_POWER_CONNECTED
01-21 03:17:18.298  3551  3551 I S.StatsService: ACTION_POWER_CONNECTED
01-21 03:17:18.326 27975 27975 D Power State: ACTION_POWER_CONNECTED
01-21 03:17:18.850 27975 27975 D Charging State: CHARGE_STOP 199044945
01-21 03:17:48.423 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 199044945 (Delay: 1000)
01-21 03:17:52.378 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 199044945 (Delay: 2000)
01-21 03:18:08.322 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 199044945 (Delay: 4000)
01-21 03:18:28.320 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 199044945 (Delay: 8000)
[... cut ...]
01-21 03:29:08.322 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 199044945 (Delay: 30000)
01-21 03:30:48.324 27975 27975 D Charging State: Fixing state w. CHARGE_ON/CHARGE_OFF 199044945 (Delay: 30000)
01-21 03:31:08.104  3551  3551 I R.System: Received intent android.intent.action.ACTION_POWER_DISCONNECTED
01-21 03:31:08.284  3551  3551 I S.StatsService: ACTION_POWER_DISCONNECTED
01-21 03:31:08.324 27975 27975 D Power State: ACTION_POWER_DISCONNECTED ignored
```

Please, reopen the issue and tell me if I can provide any help. Thanks

bam80, 2018-01-21:
@sriharshaarangi any thoughts? What else could I try? Thanks

bam80, 2018-01-21:
If it's not known yet, here is the relevant source code:
https://android.googlesource.com/kernel/msm/+/android-msm-mako-3.4-jb-mr1/drivers/power/pm8921-charger.c
Maybe it helps

bilogic, 2018-03-10:
I have the same issue on S5 DUOS G900FD, microusb charging will stop limit, but wireless charging will not stop at limit.

milux, 2018-03-12:
Sorry to say this, but I think there is little we can do at the moment.
The wireless charging is done by a different piece of hardware, and therefore control files working for USB charging may not work with wireless charging.
If you find another control file for wireless charging, BCL will still need adaption, because we have to modify 2 files simultaneously then instead of one.
Anyway, if you guys find out something interesting about the wireless charging, please let us know.

bam80, 2018-03-12:
>The wireless charging is done by a different piece of hardware

It's interesting if any of present devices has wireless charging controller capable to be managed programmatically

milux, 2018-03-12:
You nailed it... but since I don't own any wirelessly-charged device, we depend on people like you to find out.

bam80, 2018-03-12:
Well, my Nexus 4 certainly not - it's wireless charging controller pretty dumb. Not sure about others.
So it would be fine if interested people report here theirs phone or controller models

bilogic, 2018-03-13:
@milux, are there any references on how to find the control file?

milux, 2018-03-13:
Not really... have a look at the `control_files.json` to get a clue about usual places to check. However, the control files could be found in completely different places, too.

bilogic, 2018-03-13:
@milux, thanks. I had a look and also read your post on xda... and did a filter using ```battery``` what do you make of the below? Is my control file ```/sys/class/sec/tsp/cmd```? How do I test to be sure? Just modify the value of ```/sys/class/sec/tsp/cmd```?

```
D/BatteryService( 956): !@BatteryListener : batteryPropertiesChanged!
D/BatteryService( 956): level:64, scale:100, status:2, health:2, present:true, voltage: 3958, temperature: 348, technology: Li-ion, AC powered:false, USB powered:false, POGO powered:false, Wireless powered:true, icon:17303564, invalid charger:0, maxChargingCurrent:0
D/BatteryService( 956): online:10, current avg:620, charge type:1, power sharing:false, high voltage charger:false, capacity:280000, batterySWSelfDischarging:false, current_now:620
D/BatteryService( 956): Sending ACTION_BATTERY_CHANGED.
D/BatteryService( 956): Excessive delay in BatteryService : sendIntentLocked(): 23ms
D/BatteryService( 956): Sending ACTION_SEC_BATTERY_EVENT. EXTRA_SEC_PLUG_TYPE_SUMMARY : 4
D/BatteryService( 956): notify wireless on
D/BatteryService( 956): /sys/class/sec/sec_epen/epen_wcharging_mode is not found
D/BatteryService( 956): /sys/class/sec/switch/wireless is not found
I/BatteryService( 956): fileWriteString : /sys/class/sec/tsp/cmd value : set_wirelesscharger_mode,1
D/KeyguardUpdateMonitor( 1419): received broadcast android.intent.action.BATTERY_CHANGED
I/PERF ( 1419): received broadcast android.intent.action.BATTERY_CHANGED
D/KeyguardUpdateMonitor( 1419): handleBatteryUpdate
D/PowerManagerService( 956): [PSM] lowPowerModeEnabled: false (mLowPowerModeSetting: false, mIsPowered: true, mAutoLowPowerModeConfigured: false, mBatteryLevel: 64, mLowBatteryTriggerLevel: 0)
D/BatteryStatsImpl( 956): poolsize= 1, curBatteryRealtime= 2139603000, mUpdateTime= 0, mTimeout= 0, mNesting= 0, mTotalTime= 16661662000
D/UsbDeviceManager( 956): received ACTION_SEC_BATTERY_EVENT : USB CHARGING OFF
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:2 health:2
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:2 health:2
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:2 health:2
D/BatteryMeterView( 1419): onDraw batteryColor : -1107296257
D/BatteryService( 956): !@BatteryListener : batteryPropertiesChanged!
D/BatteryService( 956): level:64, scale:100, status:2, health:2, present:true, voltage: 3973, temperature: 348, technology: Li-ion, AC powered:false, USB powered:false, POGO powered:false, Wireless powered:true, icon:17303564, invalid charger:0, maxChargingCurrent:0
D/BatteryService( 956): online:10, current avg:620, charge type:1, power sharing:false, high voltage charger:false, capacity:280000, batterySWSelfDischarging:false, current_now:620
D/BatteryService( 956): Sending ACTION_BATTERY_CHANGED.
D/KeyguardUpdateMonitor( 1419): received broadcast android.intent.action.BATTERY_CHANGED
I/PERF ( 1419): received broadcast android.intent.action.BATTERY_CHANGED
D/KeyguardUpdateMonitor( 1419): handleBatteryUpdate
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:2 health:2
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:2 health:2
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:2 health:2
D/BatteryService( 956): !@BatteryListener : batteryPropertiesChanged!
D/BatteryService( 956): level:64, scale:100, status:3, health:2, present:true, voltage: 3907, temperature: 348, technology: Li-ion, AC powered:false, USB powered:false, POGO powered:false, Wireless powered:false, icon:17303550, invalid charger:0, maxChargingCurrent:0
D/BatteryService( 956): online:1, current avg:450, charge type:0, power sharing:false, high voltage charger:false, capacity:280000, batterySWSelfDischarging:false, current_now:-450
D/BatteryService( 956): notify wireless off
D/BatteryService( 956): Sending ACTION_BATTERY_CHANGED.
D/BatteryService( 956): Sending ACTION_SEC_BATTERY_EVENT. EXTRA_SEC_PLUG_TYPE_SUMMARY : 0
D/BatteryService( 956): /sys/class/sec/sec_epen/epen_wcharging_mode is not found
D/BatteryService( 956): /sys/class/sec/switch/wireless is not found
I/BatteryService( 956): fileWriteString : /sys/class/sec/tsp/cmd value : set_wirelesscharger_mode,0
D/KeyguardUpdateMonitor( 1419): received broadcast android.intent.action.BATTERY_CHANGED
I/PERF ( 1419): received broadcast android.intent.action.BATTERY_CHANGED
D/KeyguardUpdateMonitor( 1419): handleBatteryUpdate
D/PowerManagerService( 956): [PSM] lowPowerModeEnabled: false (mLowPowerModeSetting: false, mIsPowered: false, mAutoLowPowerModeConfigured: false, mBatteryLevel: 64, mLowBatteryTriggerLevel: 0)
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:3 health:2
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:3 health:2
I/MotionRecognitionService( 956): On Battery, 0
D/UsbDeviceManager( 956): received ACTION_SEC_BATTERY_EVENT : USB CHARGING OFF
D/BatteryMeterView( 1419): ACTION_BATTERY_CHANGED : level:64 status:3 health:2
D/BatteryMeterView( 1419): onDraw batteryColor : -1107296257
D/BatteryStatsImpl( 956): poolsize= 1, curBatteryRealtime= 2139603000, mUpdateTime= 0, mTimeout= 0, mNesting= 0, mTotalTime= 16661662000
I/MAXS/MAXSService( 8634): Setting status to '62-66% - Battery - 🔋 33-35°C'
```

bilogic, 2018-03-13:
I tried reading ```/sys/class/sec/tsp/cmd``` using a root browser - error opening file
Same error when trying to open it as read-write as well. Any advice? Thank you.

Perflyst, 2018-04-06:
@bilogic I cannot access this file, too. Not on the phone, not on the pc. I cannot read it over the console or with adb. I also can't copy or move this file.

But it works here without any problems. I tested it today.
Mobile phone put on charger, waited until it has reached the limit -> charge has been switched off

Same when recharging when reaching the lower charging threshold.
When I am above the upper load threshold, the app immediately intervenes and stops the load process.

I use the `batt_slat_mode`.

Phone Model: Samsung Galaxy S6
OS: LineageOS 14.1
App Version: F-Droid, 1.1.0 (22)

bilogic, 2018-04-06:
@Perflyst are you using wireless charging?

Perflyst, 2018-04-06:
Yes, I do. It works with wired charging and with wireless charging.

I do not use a case for wireless charge, it is built in my phone.

bilogic, 2018-04-07:
@milux any idea what else i can try? thank you.

Perflyst, 2018-04-07:
Another user on xda reported that it is working on his **Samsung Galaxy S8+**.
Only the battery statistics are not reset during wireless charging.

Perflyst, 2018-04-07:
Very funny, I made something there and then. Now it shows me in the notification that it is loading but unloaded in the app itself. Other battery apps behave in the same way. One says loads, the other says discharges. But I don't know if these apps detect if a power source is connected or if the device is actually charging.

My lower charging threshold is only technically maintained with the `store_mode` software when loading via usb. When I use the `batt_slat_mode` it seems to charge anyway, but stops at the upper charging threshold.

The whole thing is just a display error. The device itself (indicated by the notification LED says it is not being loaded and it always matches the settings).

Checkout the [logcat](https://privatebin.net/?73f09074832660fa#J72P9W66sGCz0PSnc8pxyey81kFR9BSxNm/6qEHF8F8=) and maybe you can learn anything from it.

@bilogic how did your logcat command look like? Could you tell me so I can enter it exactly like that and compare it with your output?

bilogic, 2018-04-07:
@Perflyst I use Battery Solo widget and the native android battery on the top right corner to determine the wireless charge/discharge status (lightning bolt = charging). As far as I can tell, the status is accurate. 

I used CatLog - Logcat Reader to extract the logs.

Perflyst, 2018-04-07:
These two outputs always appear when charging via usb or wireless.

`04-07 14:18:42.251  4928  4928 I Thread-6: type=1400 audit(0.0:553): avc: denied { open } for path="/sys/devices/battery.53/power_supply/battery/batt_current_ua_now" dev="sysfs" ino=24521 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1`

`04-07 14:18:42.251  4928  4928 I Thread-6: type=1400 audit(0.0:554): avc: denied { getattr } for path="/sys/devices/battery.53/power_supply/battery/batt_current_ua_now" dev="sysfs" ino=24521 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1`

But I can't tell if this is from Battery Charge Limit or Better Battery Stats. I don't see anything of a file like yours.

Perflyst, 2018-04-07:
@bilogic Try all three of the possible control files. I pretty much figured out how it works now. You must always restart your phone every time you make a change, regardless of whether battery stats are cleared or other charging thresholds. That's why the wrong ads came up in the previous comments. Always reboot, otherwise it won't work. If necessary, completely hibernate the app before rebooting with Greenify(root).

Gelunox, 2018-06-23:
Wireless charging works for me... sort of. It stops charging when it reaches the upper limit but doesn't restart once it reaches the lower limit. My wireless charger also stops giving charge (indicator led goes off) when charging is stopped.

However when I manually change the limit it stop when it's above the limit, and restarts when it's below (and the wireless charger also has the indicator led go off and on accordingly). So... In theory the app works with wireless charging, but in practice only half.

I'm on a Xiaomi Mi Mix 2S, control file /sys/class/power_supply/battery/input_suspend, 0, 1.

milux, 2018-06-28:
@Gelunox Is the "always write control file" option enabled? Would be worth a try.

Gelunox, 2018-06-28:
I only have "Enforce Charge Limit" enabled right now, I think always/blindly write controlfile didn't have a great effect or it didn't work. But I'll enable it and test again

milux, 2018-06-29:
Weird... can you capture a logcat during the phase when the lower limit is passed?
Would be interesting to know if the app tries to re-enable charging, or if it crashes for some strange reason.

Gelunox, 2018-06-29:
It... might actually be the charger. When I put it on my ZENS Wireless Charger Stand 10W, it starts charging again... but if I put it on my ZENS Qi Single it just unloads until I take it off and put it back on. 

Logcat I get when charging should resume: `Charging State: CHARGE_REFRESH 56790734`

Gelunox, 2018-06-30:
Yeah... it's the charger, my apologies for the wild goose chase. It works just as well with wireless charging as with wired charging for me. I'm gonna see if I can exchange the chargers that don't work properly.

milux, 2018-07-11:
I'm closing this as it is mostly about OEM-specific no-fix-stuff.

closed 2018-07-11 by milux