sriharshaarangi/BatteryChargeLimit #9: Make charge range fully configurable

blackwind, 2017-06-09:
Currently, you can only choose a lower limit of 1-10 below your upper limit. Could you make the lower limit a textbox instead of a slider so we can enter anything we want?

milux, 2017-06-09:
Actually, I thought about doing it this way. However - can you explain me a concrete scenario where this actually makes sense? Because at the moment, I don't think this is gonna happen. (Therefore the wontfix tag...)

blackwind, 2017-06-09:
My Galaxy S3 is super sensitive to charge cycles, to the extent that I have to replace my battery every 2-3 months. It's one of those scenarios where, it lasts about 24 hours at first, and by the end of 2-3 months, you take it off the charger, drive to work, and it's already lost 15% charge before you even touch it. Happens even with official Samsung batteries.

Charging to 100% and then trickle charging every time it dips below 100% (as is the default) is a major battery killer, at least on phones this old, and while charging to 90% and then trickle charging every time it dips below 80% (as Battery Charge Limit allows me to do) is certainly better, it still won't protect my battery to the extent that I need. I want to charge my phone to 90%, and I don't want it to charge again unless it's under 40%, which is still more than enough to get through the day on a fresh battery.

milux, 2017-06-20:
Full range control implemented in 0.10.0.

closed 2017-06-20 by milux