sriharshaarangi/BatteryChargeLimit #10: Battery stats reset not working on my N6P with latest PureNexus (7.1.2.r17)

aamadorj, 2017-06-23:
My N6P powered by latest PureNexus (7.1.2.r17) and rooted with Magisk v12 or Magisk v13 beta doesn't allow to reset battery stats with "dumpsys batterystats --reset" command, same as mentioned before by another Battery Charge Limit user at the XDA thread.

Newest PureNexus just added an option to reset battery stats and it seems that's done with [this commit](https://github.com/PureNexusProject/android_packages_apps_Settings/commit/48fe97486e4219912d65eedfccc1b1424f92e49e).

Can this method be added to Battery Charge Limit as an alternative to the dumpsys command?

milux, 2017-06-24:
This method seems to invoke a method `resetStatistics()` on a device-specific class `com.android.internal.os.BatteryStatsHelper` that is only available in the frameworks of this specific devices.
I might be able to call this method via Java Reflection API, will try when I find some time.
I assume that you are willing to serve as a beta tester for this, as I don't have a Nexus device? ;)

aamadorj, 2017-06-24:
Sure. I can help beta testing.

aamadorj, 2017-07-03:
Thanks for your attention to this issue.

I uninstalled release Battery Charge Limit version and tried your debug version. I really like the new interface, by the way.

I seems battery stats reset is not working yet. Logcat is attached.
[BCL.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1120622/BCL.txt)

milux, 2017-07-03:
Glad you like it. Alright, last round for today:
I did some changes with the information from your exception logs, could you please try this one:
https://mega.nz/#!N0kkGSBK!_NFwcOkJT_BgQLgqP-bDbm2S7KV935tS7NCGZdGMCK8
If this doesn't work, please provide me the new logs.

aamadorj, 2017-07-03:
The battery stats reset function of Battery Charge Limit is still not working on newest debug version.

Logcat is attached.
[BCL_2.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1120672/BCL_2.txt)

milux, 2017-07-03:
Oh boy... sorry for the trouble, but I used wrong types for the reflection API this time.
Because of insane stupidity, I give it one last try for today:
https://mega.nz/#!k4VjkJqD!cUSJFJeeFLlBQErT7RoRza5yyutrEsr2lI3b0WSXsDg

aamadorj, 2017-07-03:
No problem at all. You're doing the hard work here.

Still no luck. Logcat is attached.
[BCL_3.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1120721/BCL_3.txt)

milux, 2017-07-04:
Alright, this time it was the last instruction causing the Exception... I added something that might have been missing, let's give it one more try. Unfortunately, I'm beginning to run out of ideas.
It's really odd that I call a method "resetStatistics", with no documentation about it on the internet whatsoever... If your phone wasn't finding it, I would wonder if it actually exists...

milux, 2017-07-04:
https://mega.nz/#!kkMj3KjA!ZSuUwaWB5QRZWtvGLcBSGyFWCUzv0Z6Z5NuS1OAldQs
Make ready for the next stack trace... :wink:

aamadorj, 2017-07-04:
I really appreciate all your efforts to fix the battery stats reset, milux.

It's still not working. Logcat is attached.
[BCL_4.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1121929/BCL_4.txt)

milux, 2017-07-04:
You're welcome, but this time it might be the end of the road.
Stack trace says _"uid 10276 does not have android.permission.RESET_BATTERY_STATS"_, which is very strange, as the manifest **does** actually contain this permission.
The only assumption I have right now is that the permission is not correctly handled with an API level < 26 in the apps Manifest file.
This is bad news, because our app won't work on Android O with an API level >= 26 due to the implicit Broadcast ban of Android O.
Just for testing my assumption, please uninstall the recent beta and install this one instead:
https://mega.nz/#!Yw0BAYAJ!S39BOW9g21VcE3QKolm5cvQtlS0-82dCX5I2XFNHKlU
If I'm right, the power (dis)connect detection will not work anymore (rendering the app useless), but the reset **might** work this way.

aamadorj, 2017-07-04:
I uninstalled previous test version and installed the new one. Between two presses of the Battery Stats Reset button of Battery Charge Limit, I also connected and disconnected the USB power cord.

It seems it's all registered at the attached logcat.
[BCL_5.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1122264/BCL_5.txt)

aamadorj, 2017-07-04:
I just saw this post at the XDA Magisk v13 beta thread and I thought it can be relevant for this issue: https://forum.xda-developers.com/apps/magisk/beta-magisk-v13-0-0980cb6-t3618589/post72904730

milux, 2017-07-04:
Interesting that it still doesn't work... I don't see any of the power events in the LogCat, so my expectation about Android O seems to be correct. You can check it this way: Close the application, connect the power, and check if the notification is coming up. I guess it won't do. Anyway, I just did some experiments with Magisk 13, and it seems there are some indeed some problems.
Anyway, I don't think that the issue you pointed me to is related to the SecurityException. I have a question: This PureNexus app that you mentioned in the beginning, is it a user app or a system app?

aamadorj, 2017-07-04:
I'm in the road at this moment. Newest PureNexus battery stats function is fused with the normal Android battery information. See the screenshot attached.
[image]

milux, 2017-07-04:
Hm... let me ask differently: Is the app part of the ROM?

aamadorj, 2017-07-04:
Yes, it is.

milux, 2017-07-06:
Alright, then there is one more thing we could try:
Systemize the "normal" debug build of above (not the modified one) with the Magisk App Systemizer:
https://forum.xda-developers.com/apps/magisk/module-app-systemizer-t3477512
Then check again if it works.

milux, 2017-07-06:
Besides: The App Systemizer is officially made for up to Magisk 12... If it interferes with v13 beta... not my fault, do a backup before! 😉

aamadorj, 2017-07-06:
I "systemized" latest "normal" debug build manually with TWRP, by copying it /system/app/BCL/xxx.apk and setting the right permissions. Then I tested the battery stats reset function again two times and plugged and unpuggled the USB power cord in between.

Sadly, the battery stats were no cleared. Logcat is attached.
[BCL_6.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1127896/BCL_6.txt)

milux, 2017-07-06:
Did you uninstall any previous user version of the app in order to make sure that your systemization strategy actually works? I might be wrong (really, I don't know for sure), but I always thought that the APK has to go to /system/app directly...

milux, 2017-07-06:
And of course, if it's worth the trouble for you, you may check whether the sdk26 version shows a different behavior. You can check yourself: If the logcat contains something like this:
_New reset method failed: Caused by: java.lang.SecurityException: uid 10017 does not have android.permission.RESET_BATTERY_STATS._
then it didn't work, look for the permission error.

aamadorj, 2017-07-06:
The "systemizer" method I described is the same I use to make Greenify and other regular apps behave like system apps. I'm sure it works.

Let's wait until new Magisk is stable to see if the dumpsys issue is solved. I know the dev is aware of it.

milux, 2017-07-07:
Alright, then thanks for all your help, a pity that it didn't work. You didn't try to systemize the Android O (sdk26) build, did you?
Unfortunately, my technical understanding tells me that the dumpsys issue is an Android O issue, not a Magisk issue. Therefore, let's wait for the less experimental Android O releases, and hope that we'll see a way to cope with this issue then.

milux, 2017-07-28:
@aamadorj Starting from version 1.0.0, the app uses a new, more efficient technique to issue root commands.
Some people already reported that the statistics reset now works on their devices, and for me it also works with Magisk 13.3 and Lineage OS 14.1 (7.1.2).
Can you check if it also works for you, please?

aamadorj, 2017-07-28:
Thanks for all your work with BCL, @milux. I'll install new version ASAP.

ApertureAndroid, 2017-07-28:
It seems to work fine on other ROMs... But not PureNexus. Actually, the dumpsys command in general doesn't work in PureNexus at all with Magisk. So, now the issues with Magisk 13 we're solved in BCL and it runs the battery resetting command successfully... But PN errors out at the command.

aamadorj, 2017-07-28:
Yeap, I confirm @ApertureAndroid findings on latest BCL version: battery stats aren't resetting manually or automatically on PureNexus yet.

But the BCL enabled and BCL disabled toasts are both working now. Yay!!!

closed 2022-03-11 by aamadorj