sriharshaarangi/BatteryChargeLimit #11: Differences when using Battery Charge Limit with Magisk v13 compared to Magisk v12

aamadorj, 2017-06-26:
When I started using Magisk v13 (beta) on my N6P running lastest PureNexus, I notice Battery Charge Limit behaved differently and sometimes it didn't work properly after reconnecting.

When rooted with Magisk v12 (stable), I was used to see the "Battery Charge Limit enabled" toast as soon as I plugged the power cord in and the "Battery Charge Limid disabled" toast right after I unplugged the power cord. Also it was possible to replug the cord immediately and the enabled toast will show again.

With Magisk v13, the first toast is usually showed but the second one never shows and very often if I plug the power cord again Battery Charge Limit will not work as intended.

The logcats I capture just before I plug the power cord and right after I unplug it with each Magisk version installed are attached.
[Magisk_v12.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1100730/Magisk_v12.txt)
[Magisk_v13.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1100729/Magisk_v13.txt)

milux, 2017-06-26:
Thanks for opening an issue here!
Unfortunately, the logs don't give any hint about what might go wrong. In both cases, logs suggest successful changes as desired.
BTW: **Replugging the cord immediately is likely to fail anyway,** because our app ignores power state changes for approx. 10 seconds after the last action in order to avoid certain loops.

milux, 2017-06-26:
Could you please try again, but with a waiting time of at least 10 seconds?
Which control file is your device featuring?

aamadorj, 2017-06-26:
It seems if I wait long enough after disconnecting the power cord, Battery Charge Limit will work fine when connecting power again. I using the "battery_charging_enabled" control file with my Nexus 6P.

The only thing isn't happening with Magisk v13 is the "Battery Charge Limit disabled" toast, no matter what.

milux, 2017-07-03:
Alright. For now, let's assume that this Toast issue is some side-effect of the Magisk beta version.
I will switch to Magisk v13 as soon as it is released officially, patching any issues I might encounter then.
Meanwhile, I consider this issue as solved.

closed 2017-07-03 by milux