sriharshaarangi/BatteryChargeLimit #12: Limit charge by voltage

vovodroid, 2017-07-01:
Is there possibility to limit charge by real voltage, not by percents? It seems that percent indicator readings have no straightforward dependence with real battery voltage.
I would like to keep voltage about 3.8V, but it's difficult to say what percentage to choose.

milux, 2017-07-03:
The problem here is that the voltage will change significantly when the battery is charged.
There is no straight-forward way to determine the "idle" voltage of the battery while it is charging.
Thus, implementing this requires some heuristic to approach the desired voltage as precisely as possible.
Furthermore, the user must also provide a "recharge threshold voltage", which will certainly overstrain non-expert users.
Let's consider it a possible enhancement for an "export-mode" in a future version.

vovodroid, 2017-07-03:
> the voltage will change significantly when the battery is charged.
The simplest way is just to calculate moving average.

milux, 2017-07-03:
Maybe, but there's another problem:
The voltage also depends on the charging current, and currently I don't see any reliable way to read the latter for most devices. The higher the charging current, the higher the voltage gap.
Everything will get even worse when a fast-charging technology like QuickCharge is used.
In the end, building this feature in a meaningful way might just not be worth the trouble, if it's possible at all. Therefore, I decided to drop it some weeks ago when I thought about such a feature myself.

vovodroid, 2017-07-03:
Can at least just to voltage indicator to application?

milux, 2017-07-03:
That should be easy, as the voltage is part of the information provided by the battery intents.
Can you open a new issue for this, just for the sake of it? ;)

milux, 2017-07-10:
The information is now displayed.
If somebody is able to come up with a really good idea (or even a proof-of-concept app) for obtaining the idle voltage of a battery in a reliable manner, we can continue our discussion.

anonymouspetdo, 2017-08-21:
I found a way to limit charge by voltage on Nexus 6P 7.1.2 (N2G48B, Jul 2017).
The control file is "/sys/class/power_supply/battery/voltage_max"
The value in this file will specify the maximum voltage of the battery in mV [default value: 4400]
(Yeah, whatever the tech they use to make Li-ion go to 4.4V, battery longevity is out of the window if you leave you phone on the charger for extended time.  I lost ~600mAh out of 3400mAh by just leaving the phone on charger all the time for 2 months)

By changing this value, the charging process never goes above the specified value.

For example, the value is set to 4100.   Charging process will fast charge until the battery reach 4.1V and reduce the current to maintain battery voltage at 4.1V.  Eventually, it becomes trickle charge as if the battery is full.  Fully charge at 4.1V give battery level 80%

In my testing, there are some circumstances need to be noted.
- Since the battery voltage never goes above set value, fast charging deactivate sooner as the voltage reaches set value.  
- Unplugging&plugging the charging cable will reset the value to 4200 (Note: this is not default value).
- Reboot the phone will reset the value to 4400 (default value)

milux, 2017-08-21:
I observed a 4400 limit in some file on my device a longer time ago.
It's pretty clear that 4.4 V is **not the (idle) battery voltage**! 4400 mV is the measured voltage **at the end of the charging cycle**, and typically 100 - 200 mV above the idle voltage of the battery.
When you restrict the value of your file to 4200, I expect your battery to end up with a "real" voltage of something around 4050 mV, which would be around 70-75 % or so...
While such files are an interesting option to restrict the observable battery capacity for particular devices **_(please open another issue if you want this to be evaluated further)_**, it doesn't help with the original problem discussed here, which is:
**Find some reliable way to measure the battery idle voltage in relation to the observed voltage while charging.**
I thought about some heuristics for doing this, but so far I don't have the resources for time-consuming research.

vovodroid, 2017-08-21:
Though this limit is not idle, but end of charge volume is there chance that if keep battery constantly at this voltage it will end up with idle voltage close to this?

milux, 2017-08-21:
I think no, because a assume that's not how things work. ^^
When modifying the file in this way, this will set a new 100% mark for the charging logic of the device, however this is implemented. What the device will likely do then is something like this:
- Charge up to the new limit, let's assume 4200 mV
- Stop charging, battery voltage drops to 4000 - 4100 mV in a matter of seconds
- Wait until the battery lost some voltage due to draining or self-discharge, depending on the "intelligence" of the charging logic
- Repeat the cycle
This is nothing different from that what we have already, just another technical approach.
This approach could be more reliable or more efficient on some devices, as we don't have to mess with the charging ourselves **iff** it works. ^^
**But again, this improvement is not related to this issue, please open a new one if you want to discuss this!**

anonymouspetdo, 2017-08-21:
May I ask what kind of phone you have? What you said is not what I observed on my Nexus 6P.

Charging cycle consists of 2 phases; constant current (CC) and constant voltage (CV)
- Phase1 CC happens when the battery is low you can charge the battery with high current (2.5A for Nexus 6P) it will keep the current constantly high as long as the measure voltage across the battery is below max voltage.  If you disconnect charging cable in this stage V_idle << V_charging.
- Phase2 CV happens when the charging voltage reaches the maximum voltage of the battery.  The charging current will reduce overtime to keep the charging voltage at the maximum voltage.  Again, If you disconnect charging cable in this stage V_idle < V_charging.
- The end of charge occurs when the charging current reduce to zero in Phase2 (100% battery).
If you disconnect charging cable in this stage V_idle = V_charging = V_max (with light load <200mA. If you phone is drawing >500mA you will see voltage drop suddenly for a few 10mV)

The precise relationship of these voltages is: V_charging = V_idle + (I_c)*(R_int), where I_c is the charging current and R_int is the internal resistance of the battery.

So, the answer to this thread is: No, you cannot determine V_idle as long as charging current is greater than zero.  To determine V_idle you have to stop charging, wait 10s, and query the voltage(it's the time required for the system to update measured voltage without charging). 

On my Nexus 6P, Charging voltage reaches max voltage 4.4V at around 80% (that's when phase 2 starts. 
 If you disconnect charging cable your V_idle<4.4V). If I leave it on charging long enough (~30-50min) for it to finish phase2 (100%).  After disconnect charging cable, V_idle is actually 4.4V (With some loads it will drop to 4.37V).

For modified V_max to 4.1V, Charging voltage reach 4.1V at around 60%.  If I leave it on phase 2 long enough (~30min) the charging current will eventually reach 0 (end of phase 2) and battery is at 80%.  At this point, battery voltage will be 4.1V even the charging cable is disconnected.

The point is voltage reported during charging is not the voltage of the battery unless the charging current is very low (end of phase2).

milux, 2017-08-25:
@anonymouspetdo Interesting, thanks for that explanation. So if I understood you correctly, when we do this modification, the CV phase will already kick in at a lower voltage?
That's a surprise to me, I expected that the point of change from CC to CV was completely hardware-/voltage-driven...

milux, 2017-08-25:
@anonymouspetdo One more question: "V_charging = V_idle + (I_c)*(R_int)" is perfectly logical.
We could use this information, given 2 things:
- A file to read I_c from (those do actually exist on many devices!)
- R_int = const.

The latter might be a problem. My assumption is that R_int changes with battery temperature, which in turn changes because of environmental influence and/or charging current.
Have you done any experiments to measure the stability of R_int on your device?

anonymouspetdo, 2017-08-26:
- Yes, by modifying value in voltage_max file, you essentially change the voltage where CV phase starts.  Charging process will never use charging voltage above specified voltage in the file.
- From electrical standpoint, transitioning between CC to CV charging is not a complex process.  The circuitry responsible for charging will have a certain capability limit such as 4.3V/2A, which means you cannot exceed either 4.3V or draw more than 2A from the charging process.  Voltage from charging circuitry will drop when you try to draw current more than 2A, automatically maintain the voltage for it to supply 2A of current because it's overloaded.  When the battery is partially charged and the charging voltage reach 4.3V, the circuitry will just maintain constant voltage at 4.3V and current flow is automatically reduced.
- The internal resistance of battery changes over time, temperature, charge/discharge state, battery level, charge/discharge current.  It's certainly not a constant that you can predict easily.  It's quite misleading to separate it in the formula (my bad).  It should have been V_charging = V_batt + V(I_c;batt_state); where V(I_c;batt_state) is some additional voltage that depends on charging current and all other factors of battery state.

My suggestion is to run the service on the background (battery test mode) and monitor battery voltage during light discharge (<200mA) to determine battery voltage at different battery level percentage.  That would be quite accurate for a few ten cycles.

johnhuawie, 2017-08-31:
hi every body I need voltage limit because my p9lite dont stop charging 4.5volt andgo upper to 4.6 volt ineed ur help

Reed97123, 2018-04-17:
I would love to adjust the default value of the voltage_max file.  If this could be modified in the ROM I'm using then it wouldn't require any additional programming to limit the charging.  I also wouldn't need to root the device for the limit to function.

Where does the default 4400 get set?

Noitarud, 2024-01-02:
I'd think it's a good idea, capacitors explode because of voltage, why has this issue been ignored fo so long? Plus voltage is not uniform across the battery when it is getting dis/charged.

You could use a circuitry clamp to burn the energy, if physically opening was a better option than rooting…
Another observation I got with a phone i began reusing, two years used, four years disused, is, in about a year it began to radically decline in capacity, and i thought it was because i had left it on the charger for a while after full, which was not something i was doing when i began using again.