sriharshaarangi/BatteryChargeLimit #15: App suddenly exits

vovodroid, 2017-07-10:
Samsung S3 Android 4.3

Sometimes during overnight charge BatteryChargeLimit process exits and thus device is charged to 100%. 
Feel free to ask me any needed info, using ADB, collecting logs, etc.

milux, 2017-07-10:
Logs would be useful indeed. Some users reported similar "accidents".
See this post for instructions how to filter out relevant logcat stuff:
https://forum.xda-developers.com/showpost.php?p=72218650&postcount=581

vovodroid, 2017-07-12:
That's all:
> 07-11 21:55:49.605 D/Charging State(11919): CHARGE_REFRESH 1119214008
> 07-11 21:55:59.435 D/Power State(19944): ACTION_POWER_CONNECTED
 
In morning phone was charged to 100%, there is no app icon or process in os monitor. After reconnecting cable icon and process (actually two processes) appeared again.

vovodroid, 2017-07-14:
Yet more:

> 07-14 05:55:17.565 D/Power State(16650): ACTION_POWER_CONNECTED
> 07-14 05:55:18.935 D/Charging State(16650): CHARGE_FULL 1113997536
> 07-14 06:19:09.935 D/Charging State(16650): CHARGE_STOP 1113997536
> 07-14 06:19:14.270 D/Power State(16650): ACTION_POWER_DISCONNECTED

At  05:55 charge level was below limit, I connected cable then at 06:19 I found that charge level is above limit and still charges, so I disconnected cable.

milux, 2017-07-28:
Hey @vovodroid,
I uploaded two greatly improved versions of the app today.
I hope the changes will also fix this issue. Can you please check with latest version (1.0.1).
You can also completely disable the "state fix" in version 1.0.1, at the risk of your device ignoring the limit altogether. Check it out! :)

vovodroid, 2017-07-28:
Hi,
I didn't understand - why should I disable the "state fix"? Do you mean "Enforce Charge Limit" option?

milux, 2017-07-28:
Precisely. In previous versions, this was the source for the rapid cycling of states that you observed. If you prefer that the app shouldn't react to your device continue charging after it modified the control file, then uncheck this. Otherwise the app might try to fight with your OS, causing permanent cycling. (However, one on/off-cycle takes now one minute after a few unsuccessful approaches.)

milux, 2017-07-28:
@vovodroid: To clarify it a little better: For the first test, you should keep the "Enforce Charge Limit" in place, because it seems that your device **sometimes** continues charging, and that's exactly what it was designed for.
However, if you encounter the cycling problem again and find it annoying, you can use the option to disable the back-and-forth. The limiting will then fail silently, and your device will charge until battery is at 100%.

vovodroid, 2017-07-29:
Hi,
I run is with this option On. It was OK several cycles, but in morning phone was charged to 100%.

07-29 00:31:34.870 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 00:31:36.455 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 00:34:08.955 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 00:34:09.675 D/Power State( 8338): ACTION_POWER_DISCONNECTED ignored
07-29 00:53:12.140 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 00:53:14.000 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 00:56:06.645 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 00:56:07.570 D/Power State( 8338): ACTION_POWER_DISCONNECTED ignored
07-29 01:17:19.075 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 01:17:22.015 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 01:20:06.725 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 01:20:07.545 D/Power State( 8338): ACTION_POWER_DISCONNECTED ignored
07-29 01:41:48.130 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 01:41:50.190 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 01:44:36.280 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 01:44:37.230 D/Power State( 8338): ACTION_POWER_DISCONNECTED ignored
07-29 02:06:27.740 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 02:06:29.995 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 02:09:11.005 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 02:09:11.620 D/Power State( 8338): ACTION_POWER_DISCONNECTED ignored
07-29 02:33:28.480 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 02:33:30.565 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 02:36:08.220 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 02:36:09.190 D/Power State( 8338): ACTION_POWER_DISCONNECTED ignored
07-29 02:55:05.730 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 02:55:07.820 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 02:58:12.725 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 02:58:13.585 D/Power State( 8338): ACTION_POWER_DISCONNECTED ignored
07-29 03:26:29.090 D/Charging State( 8338): CHARGE_REFRESH 1116311112
07-29 03:26:31.045 D/Power State( 8338): ACTION_POWER_CONNECTED ignored
07-29 03:29:13.410 D/Charging State( 8338): CHARGE_STOP 1116311112
07-29 03:29:16.695 D/Power State( 8338): ACTION_POWER_DISCONNECTED
07-29 03:29:17.875 D/Power State( 8338): ACTION_POWER_CONNECTED ignored

milux, 2017-08-02:
The last line is very interesting... seems that the monitor recognized the DISCONNECTED event, but ignored the CONNECT only one second later.
I will dig into this and think about a fix. Stay tuned!

milux, 2017-08-16:
I tried to fix the issue in version 1.0.2, could you please check?

bluegizmo83, 2017-08-16:
I think I'm experiencing this rapid cycling issue on my Shield Tablet. Its charging above the limit. Went from 77% to 84% (limit set at 80%) in about 20 minutes. Filtered log attached.
[alogcat.2017-08-16-12-08-49-0600.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1229133/alogcat.2017-08-16-12-08-49-0600.txt)

vovodroid, 2017-08-17:
Hi,
1.0.2 seems much better (actually good!) now. When phone reaches charge limit, I see message "service disabled" and next "service enabled" messages and charge stopped.

milux, 2017-08-17:
That's still not exactly how it should be, but I think it's something to work with. Comment if the issue comes up again, meanwhile I close this issue.

closed 2017-08-17 by milux