sriharshaarangi/BatteryChargeLimit #16: Feature Request: ELE Phone P9000 (maybe all MTK Phones) support

kefor, 2017-07-10:
Hello, 

i think i've found the configurations named online. 

It is in /sys/class/power_supply/ac/
/sys/class/power_supply/usb/
/sys/class/power_supply/wireless/
The online file change its number for non charging 0 and charging 1

I don't know if it's important, that the online file is also available in /usb /wireless 

I also found the direct path /sys/devices/platform/battery/power_supply also with /ac / usb /wireless and the online file 

For now I wasn't able to change the numbers with a text editor, i think its the issue, that Magisk is rooting as read only.

milux, 2017-08-15:
Hey, sorry for the long delay, I'm quite busy. :wink:
We need to know if and which of this files is actually working, so you could do the following:

- Install Terminal Emulator, then execute these commands:
- su
- mount -o rw,remount <filename> (That's what BCL tries to make files writable)
- echo 0 > <filename>

milux, 2017-09-07:
Closed due to inactivity.

closed 2017-09-07 by milux