sriharshaarangi/BatteryChargeLimit #17: Info (V and T) are different in main app window and notification.

vovodroid, 2017-07-15:


milux, 2017-07-28:
I could not reproduce this issue, although it might theoretically be possible, because those updates are performed through 2 different BroadcastListeners.
Could it be that you observe this behavior exactly when your device is in **DISCHARGING** mode?

milux, 2017-08-15:
OK, since there was no feedback for quite a while, I hope that this issue is not relevant anymore and close this. Please comment if it happens again.

closed 2017-08-15 by milux

vovodroid, 2017-08-16:
> Could it be that you observe this behavior exactly when your device is in DISCHARGING mode?
Yes, it happens after charge stop, but it seems to be synchronized after some time, so I caught it by chance.