sriharshaarangi/BatteryChargeLimit #20: Support for Samsung Nexus 10 and Jiayu S3 Advance

onoakino, 2017-07-23:
Hi Sir,

Can you add support for two of my devices? 
The Battery Charge Limit says my devices are not supported yet.

On Samsung Nexus 10, rooted, using root explorer, i mounted the partition as r/w, i tried to change from '0' to '1' the value of 

/sys/class/power_supply/manta-battery/charge_enabled

Saved the file and exit.

Tried to run the Battery Charge Limit, but it still says the device isnt supported.

I then tried to reboot but the changes i made to the system didn't survive the reboot.

On Jiayu S3 Advance (please see xda), i cannot find an entry to edit using the guide you posted here. Can you point me in the right direction please?

Also, both devices are in Nougat 7.1.2 if that matters.
Jiayu is on MadOS 7.1.2
Nexus 10 is on LineageOS 7.1.2

Any suggestions?

darren3, 2017-08-12:
I would also appreciate support for the Nexus 10.

milux, 2017-08-16:
Please check out version 1.0.2 regarding Nexus support and report if it's working.

darren3, 2017-08-16:
1.0.2 didn't work for me. On XDA there is a script that looks like it does a similar thing to this app.

https://forum.xda-developers.com/nexus-10/general/script-to-control-nexus-10-charge-35-65-t3035588

Looking at it, it appears to deal with the file /sys/devices/virtual/power_supply/manta-battery/charge_enabled rather than /sys/class/power_supply/manta-battery/charge_enabled

milux, 2017-08-17:
I will have a look at it when possible, stay tuned.

milux, 2017-08-17:
I changed the control file. Please check out this experimental release if it's working: https://www.androidfilehost.com/?fid=961840155545589430
However, I also recognized that the developer of the script you linked above uses a (custom) OmniROM, that could theoretically behave differently.

darren3, 2017-08-17:
Thanks, I will give it a try. My Nexus 10 is currently fully charged so I will probably be able to let you know how I get on tomorrow. With regards the script the file is present on Lineage OS 14 so fingers crossed.

milux, 2017-08-17:
I think, because of the similarity of the paths, that it might be just the same file. Symlink or something.
But let's hope that I'm wrong. :wink:

milux, 2017-09-04:
@darren3 Any news?

darren3, 2017-09-04:
I have had some joy. When the screen is on the charge is held with the maximum I set, and doesn't drop. This is similar to what I have read on XDA happens with some Samsung phones (The Nexus 10 was made by Samsung and has an exynos CPU). I have found if the screen is off the tablet charges to 100%, I suspect this is related to doze. I have aggressive doze set when the screen is off, although the BatteryChargeLimit app is set to "Not optimised" for doze. I need to try with the screen off and aggressive doze turned off. I use aggressive doze from the app Naptime as doze is not usually supported on the Nexus 10 as I believe it lacks one of the sensors needed. I very rarely use my Nexus 10 anymore it is my kitchen tablet so don't often charge it, I will let you know how I get on without doze enabled.

darren3, 2017-09-28:
I have today updated to version 1.0.4 and put my Nexus 10 on charge. I had not altered any doze settings as I had planned and explained above. The charge is now being held to the maximum I set with the screen off, were as before this would only happen with the screen on. If I left my Nexus 10 on charge permanently now does this mean I will not damage my battery? In the past when I have left phones / tablets on charge 24 hours a day I have had trouble with the batteries swelling which is dangerous, if the battery is being held at my maximum setting is it still drawing power?

milux, 2017-10-02:
"Damage" is relative. If you don't want to "damage" your battery at all, you should keep it at ~ 65 %, which typically means something around 3.92 V, if I remember it correctly.
You may want to check the voltage directly to confirm that everything is working as it should.
Heat/swelling should not happen that much when the voltage is kept low.
I cannot say anything for your device with absolute certainty, please try it out and report errors!
Can I close this issue then?

onoakino, 2017-10-03:
Thank you darren3 and milux for continuing to discuss this. I'm following
your conversation but can't test it on my Nexus 10 right now.

milux, 2017-11-09:
Alright, report back if the issue becomes relevant again. Will close this meanwhile.

closed 2017-11-09 by milux