sriharshaarangi/BatteryChargeLimit #22: App allows full charge randomly

d0x360, 2017-08-07:
Sometimes the app will allow a full charge.  I've started opening the app before plugging in to prevent this from happening.  I wish I had more info to give.  I'm on a pixel xl running Android 7.1.1 with root.

[alogcat.2017-08-07-15-37-56-0400.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1205933/alogcat.2017-08-07-15-37-56-0400.txt)

d0x360, 2017-08-07:
I know that log isn't what you need so I will post another once my battery gets below 80%

milux, 2017-08-16:
This might be related to an issue fixed in version 1.0.2, could you please check?

d0x360, 2017-08-16:
I installed it and had the same issue.  It just kept going right past 85%.

I have what I think is just version 0.1 installed now and that works
perfectly.  Charging stops right at 85%.  I've made sure it's a protected
backup in titanium because I don't have the original APK anymore and I
downloaded the oldest version in this thread and it was different so I
think what I'm using that works is the very 1st version you out on XDA.

I'm fine with continuing to use it but the option to pick a % when it will
start charging again in the newer versions is so attractive.

Whatever the issue is I hope you figure it out but even if the problem
continues for me I have this older version that works and even though it's
something so simple I really appreciate having it. It really does work to
keep the battery healthy.

Thanks

milux, 2017-08-16:
In that case I'm sorry and hope that you can provide a new logcat that helps me figure out what's going wrong. We made so many changes since the early beta versions that the issue could basically be everything...

d0x360, 2017-08-16:
I can tell, even the settings menu in my version won't open lol but the
important bit works.

I'm try a different app to get a log.  I dont know why I couldn't get
anything when I used the filter before.  If I have any luck I'll definitely
post it.

I'll also keep checking new versions to see if whatever the issue might be
just gets fixed.  I'm on a vanilla Pixel XL so chances are good it will
just work with a future update and like I said until then I still have this
older version that works great.

milux, 2017-08-16:
For those bugs, please don't use the filter!
I should have been more precise about this, but for not strictly charging-related issues using the filter is indeed counterproductive! :wink:

milux, 2017-08-16:
And the settings not opening... sounds definitely bad... 🤔

d0x360, 2017-08-16:
On the newer APK the settings (menu in the top right corner) open fine,
it's just the old version that nothing opens and I just assumed there
weren't any at that point because it was the 1st release but I can modify
the charge value without issue so the most important part works fine.

I plan on doing a full wipe of the phone pretty soon so who knows, the
issue could be just on my end.  I've only gotten OTA updates and surely
it's rare but there's always a chance something could mess up that you
would never notice under normal use so who knows when i flash that may very
well fix it too.

I can't think of anything I've personally changed or had an app change on
the phone that would cause any issues.

milux, 2017-08-16:
Ah, got it. Yes, that was a dummy indeed. ^^
Well, if my first impression is right, the SU lib we're using since early beta version X might be causing the issues that you observe. Before, we were not using Chainfire's library, and sometimes I'm thinking this does more harm than good, especially with Magisk, which it was never designed for...

d0x360, 2017-08-16:
I'm only using chainfires su, I was considering switching but I had no
reason.  All I wanted was AdAway lol.

In the past I would be changing roms and kernels constantly but with the
pixel I just wanted a break from all the tinkering...as long as I could do
it ad free.

Once I move on from this phone I'll likely go back to constantly playing
around.  I had been doing it since the htc g1 so I just wanted a break lol.

milux, 2017-08-16:
If you are actually using SuperSU than this is even more strange... that should work perfectly with libsupersu... confusing. 😕

d0x360, 2017-08-16:
I promise I am lol.  That's it too, no Xposed, custom ROM or custom
kernel.  Just installed the updates to android as normal then I'd boot into
recovery and TWRP via adb/fastboot
, flash chainfires latest SU and that's it.

milux, 2017-09-04:
@d0x360: I had a look at the file you provided with your first post, but **it doesn't contain any trace of BCL.**
Are you sure that the app was up and running when capturing the log?

d0x360, 2017-09-12:
Yes and I know why now...

It's because I was using the app "ampere" to measure charge levels at the same time.

I'm not sure why one would effect the other but I did a full wipe and installed Oreo last week and it still wouldn't work...then I figured I'd give it a shot and found any app that is actively monitoring how much power the phone is drawing from wall will make this app not work.

d0x360, 2017-09-12:
Wish I could edit my last post.

Just want to say ty for looking into it for me, had I realized...thank you very much

closed 2017-09-12 by d0x360

milux, 2017-09-18:
That is a very interesting peace of information. Thanks for telling me!
And you're welcome!