sriharshaarangi/BatteryChargeLimit #23: Limit charge rate above/below a fixed charge threshold,or temperature etc.

tim-seoss, 2017-08-10:
Some android devices support changing the charge rate via files in /sys, slow charging prolongs battery life (and the user probably won't be inconvenienced at all for overnight charging).

Would be good to offer option to limit charge current where supported (perhaps offering option to reduce current when very empty or near target fill).

Charge rate could be automatically limited based on Android "quiet hours" setting and also device temperature (or battery temperature if available).

Charging fast towards when near full and completely empty impacts battery life (more so with some cell chemistries than others I believe).

The same goes for charging batteries which are at high temperatures or very low temperatures.  Some built-in chargers will be sufficiently intelligent to compensate for these factors on their own, but most aren't.

http://batteryuniversity.com/learn/article/charging_lithium_ion_batteries

https://www.digikey.com/en/articles/techzone/2016/sep/a-designer-guide-fast-lithium-ion-battery-charging

^^^ also see "Fast charging trade-off" section in the digikey article...

http://teslapedia.org/model-s/tesla-driver/understanding-charging-rates/

^^^ section on charging speed vs. temperature

Cheers,

Tim.

milux, 2017-08-16:
Have you been able to **actually verify** the ability of one of your devices to change the charging rate according to control file changes?
If yes, please provide more detailed information!
This could be very interesting for a future major version, but without actual experimental results I don't expect devices out there to actually support software-defined charging rates.
Greez!

tim-seoss, 2017-08-24:
The S4 i9505 (jfltexx) has kernel support for this, 

/sys/devices/platform/sec-battery/power_supply/battery/siop_level

is a writable file which can be used to set the batter charge rate as a percentage of the normal rate, this is limited to a lower bound of the USB charging rate (about 0.5 amps).

https://github.com/LineageOS/android_kernel_samsung_jf/blob/9198bbe3ef238660b18edeb4dec3790f7df06f3a/drivers/battery/sec_battery.c#L2280

https://github.com/LineageOS/android_kernel_samsung_jf/blob/9198bbe3ef238660b18edeb4dec3790f7df06f3a/drivers/battery/max77693_charger.c#L805

I've verified this with a power supply which has a built-in meter.

This may also be true for the Galaxy S3 which uses the same charge control chip (or at least the kernel could be patched to use the S4 code I suppose if-not).

It also has the ability to stop charging in software:

https://github.com/LineageOS/android_kernel_samsung_jf/blob/9198bbe3ef238660b18edeb4dec3790f7df06f3a/drivers/battery/sec_battery.c#L1077

# Cease charge:
echo 1 > /sys/devices/platform/sec-battery/power_supply/battery/test_mode

# Resume charge (and leave "test mode"):
echo 2 > /sys/devices/platform/sec-battery/power_supply/battery/test_mode

For other devices, I suppose it would be possible to limit battery temperature (e.g. only charge when temp within user-defined limits).

For devices which don't have an equiv of siop_level, you could do a basic PWM (e.g. charge for 1 second, off for 3 seconds for 25% duty), which would also AFAIK be easier on the battery than just flat-out charging...

milux, 2017-08-25:
Nice, I own an I9505 jfltexx running on Lineage 14.1 that I use for experiments.
Thanks for your help!
I will try to reproduce this and do some experiments, as soon as my time allows it

As this is a completely new thing, please don't expect this feature to become available quickly, it will take its time...

Regarding the temperature idea: How do we handle the case that the phone is too warm (maybe it slipped under my pillow while I was sleeping), but I want to have my phone charged on the next day no matter what? :wink: 

Regarding your PWM idea: This came to my mind as well. And in theory, it's a nice idea. However, in practice things like the reaction time of the device are so unreliable that we would need to do something like "1 minute charging - 3 minutes idle" for a reliable 25% duty. Changing the state within seconds might result in unpredictable outcome.

ltwiz, 2018-08-24:
`/sys/class/power_supply/battery/voltage_max` is the voltage limit file for Pixel 2 XL (and I believe other Google/Pixel devices). Interestingly this has the contents format '4400000' as opposed to some other devices simply having '4200'. Setting this limit has the benefit of battery/status automatically detecting/setting "Full" charge state (even if percentage isn't 100) and automatically toggling charge_disable to 1 - confirmed to run my device off USB power when the specified voltage threshold is reached.

Donzi84, 2022-12-01:
Hi,

I need help to understand what I have to do to block charging at 80% on a samsung tab 4

https://github.com/GalaxyTab4/android_kernel_samsung_millet/blob/millet/drivers/battery/sec_battery.c

this is the kernel link

Thank you all in advance.
Stefano

notanewbie, 2024-06-13:
I would definitely be interested in this feature, my phone stops charging all the time because it gets too hot