sriharshaarangi/BatteryChargeLimit #25: Note 3 issue

Gregbol, 2017-08-12:
Hello
My note 3 (n9005 eur)phronesis v6 phantom kernel v1.2p stops recharging normally after the app has cut the load until restart
It is as if the app remains in "no charge or discharge" mode permanently
thank you in advance

milux, 2017-08-15:
Could you provide some LogCat stuff please?
You can find instructions in the OP on XDA.

Gregbol, 2017-08-15:
https://mega.nz/#!S0xA2YYS!Hdens5FLRu0DmGSJYS86WjXNTyXQ6v-CNQl-1DUZL38
Tanks oui very much

milux, 2017-08-16:
It seems that the "Sensor Disabler" XPosed module is spamming your logs with error messages, trying to access some preference XML file that doesn't exist, you should try to fix this...
Anyway, **please provide filtered logs according to the XDA OP!**
Unfortunately, **your log does not contain any helpful information**. If BCL was running during the capture, it wasn't trying to take any action. The number of possible reasons is endless...
Are you using the **store_mode** control file on your device? **It is known to prevent any recharge until power supply is physically removed for a moment** after it has been triggered.

Gregbol, 2017-08-16:
Thank you
What should I write in the filter window.

Gregbol, 2017-08-27:
https://mega.nz/#!no4EAZzY!jvp64bG9NlRJrTwVX36cS7ECLGQEcSvxHEEM0joK42c
And with filter thanks