sriharshaarangi/BatteryChargeLimit #26: Add support for Samsung Galaxy Note 10.1

oroboros, 2017-08-13:
Please provide support for Samsung Galaxy Note 10.1 (GT-N8013).

milux, 2017-08-13:
My Galaxy Note 10.1 2014 is perfectly supported, and this post doesn't seem to provide any information about control files. Please reopen when you can provide sufficient information.

closed 2017-08-13 by milux

oroboros, 2017-08-13:
You have the 2014, which is supported. The GT-N8013 is the 2012 version, and it shows it is not supported.

milux, 2017-08-13:
Of course, but we still need information about the control file on your device, if any. We can't help without this information, as we don't have any such device for testing. 😉

oroboros, 2017-08-14:
I'm not sure what file you need.

milux, 2017-08-14:
Same here: Please read the **whole OP on XDA** and tell me if you still don't know the answer.

oroboros, 2017-08-15:
Yeah, so sorry for the confusion. I wasn't looking to collaborate. I
just wanted to try it out and saw my device wasn't supported and
wanted to let you know.

Best of luck

milux, 2017-08-15:
Well... this is GitHub, everything here is about collaboration. :wink:
You can check from time to time, maybe another device with the same control files will be supported at some point in the future.

daved44, 2024-02-19:
Milux, You commented on Aug. 13, 2017:

> My Galaxy Note 10.1 2014 is perfectly supported . . .

I have a Galaxy Note 10.1 2014, the device that has a stylus, Model SM-P600, that is dedicated to a single app to view real-time data from my weather station. It sat on a desk running full-time for a couple years, permanently charging to 100%, until recently when the battery swelled up and split the case (I knew nothing about this app until now). I will repair and continue using it, but only if I can be sure that the BatteryChargeLimit app will actually work with it. Samsung made a bunch of tablets at about that time that are similar, so I need to be sure. If you can, please tell me whether your device is or was also the SM-P600 and if so just what control file you used with the app. 

In the meantime, I have replaced it with even an older Galaxy 10.1 Tab 2, but can't get BatteryChargeLimit to function on it, so I would like to go back to the Note 2014 if I can get it to run the app and limit charge.

milux, 2024-02-20:
I'm using a SM-P605, and store mode did work nicely for me with stock. However, I'm using lineage for years now (Android 7), and there at least store_mode is somewhat working, keeping the battery around 80 %...

daved44, 2024-02-20:
I did a bit of research this morning and it appears that these two device models are similar with mine doing only WiFi and yours also doing cellular, but with the same chip sets and released at the same time. But that probably isn't enough to ensure that these two devices are identical enough to have similar basic ROMs. I may make the plunge, but there are enough tablets from that era available on eBay that maybe I'll buy one and root it at whatever level it is instead of trying to repair mine. 

I would still like to know what configuration file you are using. Thanks for the quick response.

milux, 2024-05-10:
Last comment on this: Recognized that with my Lineage ROM, `/sys/class/power_supply/batt_slate_mode` is/was the way to go.
`store_mode` worked great on stock, but under Lineage it prevents any battery charging after being triggered for the first time, until the device dies from low battery or is manually rebooted.