sriharshaarangi/BatteryChargeLimit #29: Support for Xiaomi Mi6 - I found proper file to enable/disable charging

Shackmir, 2017-08-15:
I foundright file which works for Mi6. 
by editing this file

"file": "//sys/class/power_supply/battery/input_suspend",
"label": "input_suspend",
"details": "Xiaomi Mi6",
"chargeOn": "0",
"chargeOff": "1",

I tested it and works without  any problem, please add support for Mi6 ASAP. Thanks

milux, 2017-08-16:
Please check out version 1.0.2 and report if it's working.

Shackmir, 2017-08-16:
It works. Disables and enables charging automatically. App crashed few times but I whitelisted it in battery saving settings and now it works flawlessly. Thank you.

Shackmir, 2017-08-16:
I found files where charging current can be modified. Is this enabled in this app on other devices? I can provide directory of this file.

milux, 2017-08-16:
Thanks.
No, there isn't any feature like this yet.
If you have actual experimental results about software-define changes of charging current, please comment at https://github.com/sriharshaarangi/BatteryChargeLimit/issues/23

closed 2017-08-16 by milux

Shackmir, 2017-08-16:
Unfortunately wiping battery stats on Xiaomi mi6 doesn't work.

milux, 2017-08-16:
That's not related to the control file, please open a new issue for this.