sriharshaarangi/BatteryChargeLimit #30: Smartmanager reports 12 fc's in the last 5 days

twinko, 2017-08-16:
Hey following setting:

-S7 edge (exynos) 
-Nougat (batman rom)
-magisk 13.5 rooted

The samsung smartmanager reports 12 force close of bcl in the last 5 days. Beside that its working 100%. I mean the app does what it should do. But im scared it could drain battery or something like that if its fc's and needs to restart.

If i need to provide a log please give short advice how to do that. If you give me the codes im able to extract necessary information via adb or something else :)

Regards

bluegizmo83, 2017-08-16:
Logs are always helpful. Scroll down to "Method 2 aLogcat App", about half way down the page here, easy to follow directions: https://android.gadgethacks.com/how-to/android-basics-capture-logcat-for-detailed-bug-reports-0167774/

Note: if you can't reproduce the Force Close on demand, you may need to start aLogcat and Grant superuser the first time you open it, then close it and wait until you notice the Force Close count increase in Smartmanager. Then open aLogcat again and save the log, the error should be in the log then.

bluegizmo83, 2017-08-16:
There is also instructions for logs (including how to filter the logs down to relevant information, which is preferred) in the first post of the xda thread for this app, under the 2nd question of the FAQ.
https://forum.xda-developers.com/android/apps-games/root-battery-charge-limit-t3557002

milux, 2017-08-16:
Would be good to know what's causing the FCs. Please provide your logcat **without any filters!** I need a stack trace of a crash or something.

milux, 2017-08-18:
Just wanted to add that the app is entirely inactive when there is no power supply.
So if battery drainage is your only concern, I can immediately close this. ^^

twinko, 2017-08-18:
im sorry, i need to hand in my masters thesis on monday, will come back to you next week with some logs :)

milux, 2017-08-18:
Hehe, I know that situation quite well. Wish you all the best! What is it about?

twinko, 2017-08-18:
the question is: does the rais of structure inside of an growing company has an impact on the motivation level of there employees. :)

milux, 2017-09-07:
@twinko Any news?

twinko, 2017-09-07:
@milux i prepaired everything but it looks like the error is not appearing anymore. Smartmanager keeps quiet and is not reporting any errors. Moreover im not able to look into an fc counter for apps or something, it was only displayed in this specific smartmanager popup.

milux, 2017-09-08:
No, I think that's not necessary.
During the last months I fixed one or two issues that could cause NPEs, and more recently (release 1.0.3 I think) I migrated the whole app from Java to Kotlin, which has built-in Null-Pointer-Safety (with few exceptions because of interfaces to Java code).
I think that we're good. Thanks for your answer, I close this.

closed 2017-09-08 by milux