sriharshaarangi/BatteryChargeLimit #32: Shield Tablet rapid cycling...?

bluegizmo83, 2017-08-16:
I believe the I'm having the rapid cycling issue (issue #15 ) on my Shield Tablet. I plugged it in several hours ago at 60%, came back to it just now and it was at 97% (all app setting at default, including 80% limit). Attached is a filtered logcat.
[alogcat.2017-08-16-15-47-48-0600.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1229625/alogcat.2017-08-16-15-47-48-0600.txt)

milux, 2017-08-17:
You're right, except it's not rapid, because of the exponential back-off I implemented.
If the app behaves in this way, it's usually because the device doesn't react to the changes of the control file and continues charging.
Could you provide a logcat of the events **before** the cycling starts? You can set your limit one percent above the current level and enable, this should reproduce it.
Please also try to set the limit **below your battery level** and see if the app does actually stop charging when you toggle the service on/off.

bluegizmo83, 2017-08-17:
Ok I'll give that a shot. I'm not sure if your familiar with the app Automate? It's a Tasker style app. I created a flow in that app for my Shield Tablet that does basically the same thing that your app here is trying to do, it watches the battery percentage while the charging cable  is connected and disables charging when the battery percentage is >= 80%, by editing the control file, and it works perfectly. Once it disables the charging through the control file edit, it stays disabled and the battery percentage stays right where it's supposed to, so I don't THINK it's specifically a device/control file issue per say... Maybe I'm wrong?

milux, 2017-08-17:
When you look on the first pages of the BCL XDA thread, you will find that I did quite complex Automate flows for exactly the same reason before Harsha published the first version and I started to support him with development of the native app.
So yes, one could say that I'm quite familiar with Automate. :wink:
Please send me your flow file, I will check it out. But nevertheless, I will need your support to check what exact difference between your flow and our app causes the wrong behavior.
This app is so extremely hard to develop because I don't have physical access the the devices I have to fix. :wink:

bluegizmo83, 2017-08-18:
Here is a copy of the flow I created, i had to zip it cause I couldn't attach a .flo file here (its also available on the automate community). I double checked by testing it again all night last night, woke up this morning my tablet was still right at 80%.

[Charge Limiter (ROOT, Shield Tablet).zip](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1234607/Charge.Limiter.ROOT.Shield.Tablet.zip)

bluegizmo83, 2017-08-18:
You'll notice in my flow, there is no re-enabling of the control file. I'm thinking that maybe on some devices (maybe even most?), when you disable the control file, that basically cuts off all input current from the connected charge cable. So you need to keep checking charge level and re-enabling the control file ever so often to maintain a certain charge percentage. On the Shield Tablet though, when I disable the control file, that doesn't cut all input current. The device basically maintains a 0 mAh level (no negative mAh drain, and no positive mAh charge). When I disconnect the cable anytime after it's reached the charge cutoff, I also do not need to re-enable the control file in the flow on this device because as soon as the cable is disconnect the system automatically changes the control file back to "enabled", so I never re-enable the control file in the flow.

milux, 2017-08-18:
I noticed that, and I think you might actually be right.
What your hardware could do different is that it might still report "CHARGING" although it doesn't charge the battery anymore, just keeps a balanced in/out ratio.
If you're right, the issue is **really very easy to fix**.
Just go to the **settings** and **disable the "Enforce Limit" option**. I implemented this option for exactly such cases. The app should then just let the device "do it's thing" after it changed the control file.

bluegizmo83, 2017-08-18:
Yup I'm pretty sure that's exactly what's it's doing. I remember now seeing in BCL that it said "CHARGING" in green before and after it reached the limit. I didn't even think to disable the Enforce Limit option! I'll definitely give that a try and let you know. I know a little bit about programming, I taught myself some Visual Basic back in the early 2000's when I was a teenager, and have done some hobby work with Arduino. I'm no expert by any means though, I know just enough to be dangerous, as they say lol.

milux, 2017-08-18:
I guessed. :wink: Had a look at your flow. The delay is inefficient, you should use another battery block with "When" (continue on change instead of immediately). When you re-arrange the other blocks to look less messy, this flow is a really good one!
Never did anything with Visual Basic. (Honestly, I think it's an ugly language. PHP causes me enough headache for the rest of my lifetime, don't want to cope with just another shitty language.)
I know Java, Python, C, C++, Node.js and all the web language stuff quite well. Finished my MSc Informatics last year, so I think it's fair when I have a bit more knowledge. ^^
At the moment me and a friend of mine are doing hobby experiments with multiple Arduinos that communicate over HC12 modules. It will be kind of a DIY alarm system with invisible laser barrier. Just for fun. :wink:

bluegizmo83, 2017-08-18:
Thanks. I never claimed my flow wasn't messy lol 😄 
Java is the one language I really wish I'd have learned, it's used it so many things, from Android apps, to routers, to IofT devices, just about everything uses Java in some form or another these days.

milux, 2017-08-18:
Java on IoT❓ Not a bad idea in principle, but especially for smaller devices it's typically C(++) with all the nasty security consequences... And the routers I've seen do typically run on C/C++ mixed with all kind of scripting languages like Lua or PHP. But maybe some companies use Java, too...

bluegizmo83, 2017-08-18:
Disabling the Enforce Limit tick box did the job! Staying right at 80% now! Maybe set the app to disable it by default on the Shield Tablet, so future users won't have to worry about it.  thanks again!

milux, 2017-08-20:
I will consider changing the JSON-Format of the devices list file to include a flag for this. Thanks for reporting, issue closed. :)

closed 2017-08-20 by milux