sriharshaarangi/BatteryChargeLimit #33: Shield Tablet random crash after unplugging

bluegizmo83, 2017-08-16:
I received a random crash at 17:29 today after unplugging power cable. Attached is an unfiltered logcat. If this is another weird crash like that crash i had at first run after installing the app this morning, don't worry about it, just close the issue. I'm going to uninstall this app and wait to test again until I get my system straightened out from that bad Titanium Backup restore I did a while back. I'm not really helping the situation here by continuing to test with a known system issue on my end. Thanks for your work on this app and adding experimental Shield Tablet support though! I appreciate it.

milux, 2017-08-17:
I think I fixed it. However, a "normal" user won't encounter it normally, because it only occurs in combination with the cycling problem you described in the other post.
I will therefore close this issue, let's focus on the cycling, that's the real problem.

closed 2017-08-17 by milux

milux, 2017-08-17:
You can find a corrected experimental build here:
https://www.androidfilehost.com/?fid=961840155545589430
This one shouldn't crash anymore when the power plug is removed. If it still does, I screwed up. 😉