sriharshaarangi/BatteryChargeLimit #35: Support for Nexus 9

zycho42, 2017-08-23:
I found the following on my nexus 9:

flounder:/ # cat /sys/devices/platform/tegra12-i2c.0/i2c-0/0-006b/charging_state
enabled

Can you add support for the Nexus 9? It would be much appreciated!

milux, 2017-08-25:
Have you tried to modify it with a su shell to see if your device reacts properly?
Which string needs to be written to disable charging ("disabled"?) and to resume ("enabled"?)?

zycho42, 2017-08-26:
Yes I have!
After plugging in I executed:
`flounder:/ # echo disabled > /sys/devices/platform/tegra12-i2c.0/i2c-0/0-006b/charging_state`
I was sceptical at first because the battery icon still showed it was charging 'lightning bolt' in the battery. But after an hour the percentage was the same. I then performed:
`flounder:/ # echo enabled > /sys/devices/platform/tegra12-i2c.0/i2c-0/0-006b/charging_state`
And after 30 minutes the percentage had increased 4%. 
So:
to disable charging: "disabled"
to resume charging: "enabled"

Thanks!

milux, 2017-09-08:
@zycho42 Your device should be supported by now, could you check and report?

sriharshaarangi, 2018-05-15:
Since, there is no response, closing this. The app should be working on the device. Reopen in case it doesn't work.

zycho42, 2018-05-15:
Sorry for not responding. I unrooted and totally forgot about this thread!
No need to reopen on my account.

tadly, 2018-05-16:
Well that's bad timing on my part I guess?
Rooted my N9 the other day just for this and it kind of doesn't work.

I haven't tested the above yet but will do later today and report back.

tadly, 2018-05-17:
So... a day late but here I go.

Tested what @zycho42 did back in August 2017 with the same result.
`disabled` and `enabled` work when doing it manually.

Via the app it doesn't.
Reading the value of `/sys/devices/platform/tegra12-i2c.0/i2c-0/0-006b/charging_state` shows it's always set to `enabled` even if it should be `disabled`.

I'm running a stock version of Android 7.1.1.
Anything else I should test?

p.s. Wasn't there a way to re-open issues on github or am I going mad?

sriharshaarangi, 2018-05-17:
Strange. @milux Any ideas?

tadly, 2018-05-29:
Um so, any news on this?
I still don't even know if `/sys/devices/platform/tegra12-i2c.0/i2c-0/0-006b/charging_state` should actually change (I'd assume so but you never know).

Anything I can to do get this resolved?

milux, 2018-07-11:
@tadly Sorry for the loooong delay, my calendar is awfully packed these days...
You said that the file "shows it's always set to enabled even if it should be disabled".
If this is so, then the app will surely misbehave in one way or the other. What happens when you enable the "blind writing" in settings? (It was actually implemented for such cases.)

tadly, 2018-07-16:
@milux sorry for the delay too.
So I managed to get the app to do what it's supposed to and I verified that `charging_state` was set to `disabled` while having it connected to my pc.
After hooking it up to the wall-plug, it started charging again even though it shouldn't at which point I didn't feel like investing any further (to lazy to set up adb over tcp).

I also got myself a Lenovo Tab4 now and making the Nexus 9 obsolete...

apeczeli, 2018-09-11:
Hello,
I have the same equipment with the same issue. It doesn't stop charging at 80% even though the app seems to work fine. I have attached a logcat log of the event of inserting the power cord when battery is over 80%. Please can you inspect it and help me to make it work? Thank you.
[alogcat.2018-09-11-08-34-13+0200.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/2369627/alogcat.2018-09-11-08-34-13%2B0200.txt)