sriharshaarangi/BatteryChargeLimit #37: Repost note 3 issue with store_mode

Gregbol, 2017-08-27:
https://mega.nz/#!no4EAZzY!jvp64bG9NlRJrTwVX36cS7ECLGQEcSvxHEEM0joK42c
Here is a link with filtres.
My phone not charge after stopped by the app.
Until next reboot.
Thanks

milux, 2017-08-31:
Is there any other control file supported on your device? **store_mode** is known for a **variety of strange behaviors**, most of them not fixable.

Gregbol, 2017-09-01:
Hello.
How is the report.
Is everything normal or there are unknown errors?
thank you

milux, 2017-09-04:
The file contains 4 events:
- Power supply was connected @ 15:40:16
- App starts monitoring charging @ 15:40:17
- App stops charging @ 15:42:01 (Likely because the device has reached the set threshold)
- Power supply was removed @ 15:42:29 (Either a very late reaction to the previous change or you removed it physically.)

Looks pretty normal to me, can't see anything unusual. So what exactly was the problem you were experiencing while taking this log? 😕

Gregbol, 2017-09-04:
It is only after the application stops the load at 80% eg if I disconnect the charger and discharges the battery up to 75% it is impossible to restart the charge the phone remains in charge maintenance mode it does not charge or discharge until the next reboot.

milux, 2017-09-05:
Well, looks like a device-specific relative of the other known issues with **store_mode**.
Let me ask you one more time, because afaik you still didn't answer this question:
**Is there any alternative control file supported for your device?**

Gregbol, 2017-09-05:
Thank you for your attention.
I do not understand what you mean by control file.
But I did some research, asked info tried to change to 1 the 0 of some files in sys / class / power_supply / battery /
without success.

milux, 2017-09-06:
I mean when you open the settings (menu bar) and tap the very first option (control file), is the any other item selectable except store_mode?

Gregbol, 2017-09-06:
Yes there is the "batt_slate_mode" but this one does not stop the charge the phone charges up to 100%

milux, 2017-09-07:
OK, usually it does, but with different semantics... batt_slate_mode works fine on 2 of my devices.
Have you tried what happens if you discharge further - let's say down to 50 % - and then try to recharge?
Did this change anything, or is it entirely impossible to charge once store_mode was modified?

Gregbol, 2017-09-07:
I will try thank you
but how do I change store_mode

Gregbol, 2017-09-08:
Thank you for your help mieux.
It seems that with the latest kernel phantom 1.2 s beta the software works in batt_slate_mode and it is a real relief
A +

Gregbol, 2017-09-08:
@milux not mieux and thank you again.

milux, 2017-09-08:
You're welcome, glad that you could find working parameters.

closed 2017-09-08 by milux

Gregbol, 2017-09-08:
I tell you?
the app does not work anymore when I turn off the phone the battery goes up to 100 and when I use it plugged it decends very fast a 78 then goes back up little by little.
It's not very important I guess someone will find a solution in a while.
Let's wait and see.
A +

milux, 2017-09-18:
When you turn off the phone the watching service cannot run, therefore there is no control.
This cannot be fixed in any way.

Gregbol, 2017-09-18:
Thank you
I solved the problem
I give the value 1 a batt_slate_mode whenever the battery changes and between 85 and 100%
and it works