sriharshaarangi/BatteryChargeLimit #38: Support for Nexus 6 please >:o)

ottomanmint, 2017-08-29:
I’ve found the following file /sys/class/power_supply/battery/charge_enabled on my rooted Nexus 6 (Shamu) & toggling 1 to 0 does stop charging the device. The same file is selected by default in your apk.
Can you please add support for the Nexus 6?

Issue: The latest version of your apk does not function as intended on my N6 running RR Unofficial Android 7.1.1. (I gave it SU rights).
The Battery Charge Limit apk seems to stop the charge at the set limit once, however the N6 will not charge again once below the lower limit set in the apk unless rebooted. Than it will charge to the set limit again once only! I also don’t see the value of 1 change to 0 in the above mentioned control file.
I also don't like the idea of the phone being kept awake always if it is on hte charger...
Couldn't the ROM be modified to se 75% as full?
Let me know if I can help you in any way. Thank you!
Cheers

testman86, 2017-08-29:
Excact same problem here, as described in the thread -> https://forum.xda-developers.com/showpost.php?p=73588529&postcount=996 
LineageOS 14.1 25th August build, Android 7.1.2.
Would be really interesting why it worked before (maybe I did reboots without noticing them cause I was setting up my device from scratch two days ago.)

testman86, 2017-08-29:
I played around a bit and after a
"echo "0" > /sys/class/power_supply/battery/charging_enabled"
in termux with root privileges its working properly again without an reboot and BatterChargerLimit directly pops-up again and says "Waiting till 70%".
So something is not properly switch back the state to 0 after full charge.

I also recognized, after charging, and waiting till 3% dropped (to 67%) i can still charge again, even if my lower threshold is 60%. So somehow its starts charging again even if it shouldnt.

testman86, 2017-08-29:
And most interesting. After 
"echo "0" > /sys/class/power_supply/battery/charging_enabled"
if I do a
"cat /sys/class/power_supply/battery/charging_enabled"
its still at 1 and not back to zero, but now it can charge again without a reboot.

testman86, 2017-08-29:
and after a reboot a 
"cat /sys/class/power_supply/battery/charging_enabled"
gives also a 1

testman86, 2017-08-30:
And this morning (aqfter reboot yeseterday) it charges till 70% limit as it should.

a "cat /sys/class/power_supply/battery/charging_enabled" returns a 1 afterwards BUT if i plug in the power supply again (and beeing above the threshold) it still returns a 1 and its NOT charging.

I would exspect that BatterChargeLimit set it back to zero so that charging is disabled. Its set on 1 (so charging) but phone is still not charging.

milux, 2017-09-04:
I think that this issue **might** be caused when your device performs changes to the file without altering its behavior in the expected way.
**Please try the "always write" feature of version 1.0.4 and report back.**

ottomanmint, 2017-09-09:
Hi,
Sorry about the late reply...
Thanks for addressing the issue, but it did not work out as planned.
Currently using a software with audio warning for low & high charge limit
set. Of course I can only monitor the charge during my up time =)
Will look into aLogcat usage etc.
Cheers

ottomanmint, 2017-09-10:
Hi,
Tried again with always write & the power option as well in settings, but
no luck.
Cheers

testman86, 2017-10-01:
Sorry for the late reply. I was testing the behaviour in longterm. After reboot, or if not sharing till the threshold and unplug the charger before reaching the upper threshold everything works "fine". But waiting till upper threshold will prevent further charging till reboot or a skript to enable booting again.

sriharshaarangi, 2018-05-15:
@ottomanmint @testman86 
Can you observe any changes with the v1.1+ versions? Or is it behaving in the same way?

milux, 2018-05-15:
@testman86
In case it's not working still, could you please provide a link to the "script to enable charging again"?
This might be helpful for us to understand possible workarounds.

ottomanmint, 2018-05-15:
Hi,
Installed from apk store & charged my Nexus 6 from 38% to the set 45% as
expected & stopped charging!
Now at 38% discharging, however the Battery Charge Limit apk is flaming
that it is charging to 45% again. Disconnect & reconnecting the charger
does not help. I am sure rebooting would, just like before.
Where is this script located again?
I'll try it on my Nexus 10 later...
Cheers

testman86, 2018-05-22:
@milux i use termux and the termux:widget for it. In the script I have a one line code snippet:

"
su -c 'echo "1" > /sys/class/power_supply/battery/charging_enabled'
"

this enables charging again. I charge till the upper limit, unplug the phone and use it as long as I want. Then plug it in again to the charger, run this line of code and the phone charges again and stopps after reaching the upper limit. Without running this code a reboot is needed before the phone will charge again.

digitalcircuit, 2018-10-27:
In the latest release of Battery Charge Limit (v1.1.1), enabling the `Always write CTRL file` option appears to make this app correctly start and stop charging on my Nexus 6, even Qi wireless charging (to my surprise).  LineageOS 15.1, Android 8.1, with Magisk, just in case that matters.

Without that option enabled, the same symptom arises that charging is disabled, but not re-enabled.  I haven't tried watching the `charging_enabled` file to see if it somehow gets reset, misleading Battery Charge Limit.

ottomanmint, 2018-10-27:
Hi,
I've confirmed it works on my N6 great thanks!
I wish I knew how to tinker with apk's, however I barely find time for my
self lately...
However after install on my N10 puts the tablet into a boot-loop still! I
have confirmed this by using a second marshmallow rom as well DU & RR!
After the installation the tablet screen does not respond for a few seconds
& then the boot-loop begins...
Cheers

milux, 2018-11-05:
> Hi, I've confirmed it works on my N6 great thanks! I wish I knew how to tinker with apk's, however I barely find time for my self lately... However after install on my N10 puts the tablet into a boot-loop still!

Good to hear. Sounds like N10 does not like the commands of BCL. There's likely nothing we can do for this on our side. If you find something that we can do, please open another issue for this!

milux, 2018-11-05:
> @milux i use termux and the termux:widget for it. https://termux.com/add-on-widget.html
> In the script I have a one line code snippet:
>
> "
> su -c 'echo "1" > [...]

Have you tried the solution of @digitalcircuit ?