sriharshaarangi/BatteryChargeLimit #39: Support for Nexus 10 please =o)

ottomanmint, 2017-08-29:
Thank you for your work on the Battery Charge Limit apk! I know there is some support for the N10 already...
Issue: It does not function as intended on my N10 running DU Android 6.0.1! It charged up to the set limit the first time & the next time I put my N10 on charge (once the charge level was below the set low limit) my N10 it got fully charged the second time around, despite the limit been set to 75%.

milux, 2017-09-05:
A logcat, captured during the time when the issue occurred, would be helpful. Please see XDA OP for instructions.

ottomanmint, 2017-09-09:
Hi,
I will look into aLogcat usage etc.
Cheers

sriharshaarangi, 2018-05-15:
@ottomanmint 
The app was crashing sometimes due to some bugs, leading to full charge. Many of them are fixed in v1.1+. Can you test the latest version and see if it is working as expected?

ottomanmint, 2018-05-15:
Hi,
Installed from apk store & charged my Nexus 6 from 38% to the set 45% as
expected & stopped charging!
Now at 38% discharging, however the Battery Charge Limit apk is flaming
that it is charging to 45% again. Disconnect & reconnecting the charger
does not help. I am sure rebooting would, just like before.
Where is this script located again?
I'll try it on my Nexus 10 later...
Cheers

milux, 2018-07-11:
@ottomanmint Just wanna mention that no code is run during the install process of the APK. Therefore, your paranormal softbrick-experience is not caused by our app.
Closing this now, as you placed all the relevant information in the other thread already.

closed 2018-07-11 by milux