sriharshaarangi/BatteryChargeLimit #40: support for teclast x98 air lll (M5C5)

laptapper1, 2017-08-30:
Congrats for a good job. Your APP is working perfectly in my mobile phone.Is it possible to support the teclast x98 air ? Which Informations you need oft the Intel cherry trail tablet?
Some files are located /sys/class/power_supply/dollar_cove_charger/enable_charging
Thanks in advance!

milux, 2017-08-31:
Please report what date has to be written to the reported control file

1. to disable charging
2. to re-enable charging

laptapper1, 2017-09-01:
Thanks for the fast reply.
It looks like 0 for disable charging and 1 for enable charging.
I have tried to change it manually with es file Explorer but I have no writing rights in sys (I am wondering because it is a rooted tablet)

laptapper1, 2017-09-02:
[image]
[image]
[image]
[image]
[image]
[image]

I'll hope that my pictures of all available power folders are helpful.
I would be glad if you're able to get your perfect app working with teclast tablets.
Thanks

laptapper1, 2017-09-03:
I've found out that I can stop the charging with these commands:
chmod 4777 /sys/class/power_supply/dollar_cove_charger/present
su
echo "0" >/sys/class/power_supply/dollar_cove_charger/present

But I can only charge again when I restart the UI or unplug and replug the power cable physically.
hopefully it helps....

laptapper1, 2017-09-03:
Hi 
I've send answers directly in github

milux <notifications@github.com> schrieb am 31.08.2017:
>Please report what date has to be written to the reported control file
>
>1. to disable charging
>2. to re-enable charging

milux, 2017-09-04:
Experimental support will be added with upcoming release, but I can't tell whether the problem with the reset of charging will also affect our application.

laptapper1, 2017-09-06:
Hello
Thanks I've tried the new app but it isn't working. It seems to be a write protection problem. The path is ok. With tasker I got it working but only with chmod changing present. Please could you check this in your app. Thanks

milux, 2017-09-07:
please see issue #46

laptapper1, 2017-09-08:
But the problem is that if I restart the tablet the write protection is back to write protect

milux, 2017-09-08:
Therefore I will have to implement a check that executes chmod when necessary, as discussed in #46.

sriharshaarangi, 2018-05-15:
@laptapper1 Can you try the latest release v1.1.1, and report if it working as expected?
Write protection is fixed here: #92 
The control file is present as well.