sriharshaarangi/BatteryChargeLimit #43: Support for Xiaomi Redmi Note4/Note4x

Moulick, 2017-08-31:
Model: Xiaomi Redmi Note4/Note4x
Android version: 7
Miui version: 7.8.24

There are two control files for changing charging state.
1. sys/class/power_supply/battery/battery_charging_enabled
2. sys/class/power_supply/battery/charging_enabled

For both files, 0 is not charging, 1 is charging
Changing the value of either of the files turns off the charging. Even if any one of the files have value '0', charging is cut-off.

I'll be happy to provide any other info

milux, 2017-08-31:
I don't understand the reason for this issue.
Both files you reported are supported since the early 0.x times?
From the JSON file for control files:
```
{
  "file": "/sys/class/power_supply/battery/battery_charging_enabled",
  "label": "battery_charging_enabled",
  "details": "Nexus/Pixel",
  "chargeOn": "1",
  "chargeOff": "0",
  "experimental": false
},{
  "file": "/sys/class/power_supply/battery/charging_enabled",
  "label": "charging_enabled",
  "details": "Generic",
  "chargeOn": "1",
  "chargeOff": "0",
  "experimental": false
}
```

Moulick, 2017-08-31:
The app is not stopping the charging. Neither of the control file worked in the app. I have set the limit to 85%, but the charging didn't stop.

milux, 2017-08-31:
I need an **unfiltered** logcat of our app failing on your phone which helps me to evaluate the issue.

Moulick, 2017-08-31:
How should I go about it. I have installed aLogcat.

Moulick, 2017-08-31:
okay, here is the logcat. 
[alogcat.2017-08-31-17-57-53+0530.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1266995/alogcat.2017-08-31-17-57-53.0530.txt)

milux, 2017-09-04:
Sorry, but in your log **there is no log message of BCL at all.**
Are you sure that BCL was running at the service was enabled when you were capturing this?

Moulick, 2017-09-04:
Yeah, BCL was running. I'll capture and provide another in just a minute.

Moulick, 2017-09-04:
[alogcat.2017-09-04-18-47-39+0530.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1274788/alogcat.2017-09-04-18-47-39.0530.txt)

Okay, I connected the charging at around 82% and let it charge till 87%. The limit in the app is set at 85%

[image]

[image]

milux, 2017-09-04:
OK, I see relevant messages on lines 366, 440 and 564, but I cannot say anything for sure because **your log is missing timestamps**.
BCL tries to stop charging in line 366, recognizes that power supply is off (likely because of the control file change) in line 440, and then reports that power was attached again in line 564.
The last one could be problematic if I knew the timing. It seems that your device resumes charging some time after BCL changed the control file. Have you enabled the "enforce limit" option in settings?

Moulick, 2017-09-04:
I am using the aLogCat app to get the log, I can't find any setting to enable/disable timestamps. Is there any other app that I should try to get the log?
The enforce limit option is enabled in the settings.

milux, 2017-09-04:
There's an option for the **log format**. Please see the OP on XDA, it's all there...

Moulick, 2017-09-04:
Yeah, so the app is successfully stopping charging now. The problem was the advanced setting 'Always Handle Power Events'. Disabling it made the app work.

milux, 2017-09-05:
Ah, I see... this settings defaults to "false" for a good reason.
It can prevent unnecessary delays on devices with certain control files, but for most control files a "power off" event is fired directly after disabling the charge, and this has to be ignored.

closed 2017-09-05 by milux

Moulick, 2017-09-05:
Yeah, I don't know how it flipped. Must have enabled if accidentally. Thanks for troubleshooting.