sriharshaarangi/BatteryChargeLimit #44: I need volt charging limmter

johnhuawie, 2017-09-01:
hi every body I need voltage limit because my p9lite dont stop charging 4.5volt andgo upper to 4.6 volt ineed ur help  percentage is always false  hardware problem but give real battery volt

milux, 2017-09-04:
This has been discussed as being not practically possible at the moment. Please look for the relevant issues and join the discussion there.
You might also try to ask a little more friendly, this will increase the motivation of the developers. :wink:

closed 2017-09-04 by milux

johnhuawie, 2017-09-04:
thx for u but I charging my phone by looking to voltage when it become 14.4
I stop charging I need only ur app doing that for me or any simple app

johnhuawie, 2017-09-04:
I read all ur chatting about volt limtter and ur right good charging is to read percent but I say if u can do that for me