sriharshaarangi/BatteryChargeLimit #45: Nexus 5 support

fredsilon, 2017-09-03:
_Nexus 5 running LineageOS 14.1-20170829-NIGHTLY-hammerhead_

Control File (default = read-only ... -r--r--r--):
**/sys/class/power_supply/ac/charging_enabled**
1 = charging enabled
0 = charging disabled

I had to change the file to read/write:
`/sys/class/power_supply/ac # chmod u+w charging_enabled `

**Testing**
```
/sys/class/power_supply/ac # echo 0 > charging_enabled
/sys/class/power_supply/ac # cat status
Not charging
```
Waited for 10 minutes ... battery charge level didn't go up.

After setting the file back to 1, the phone got charged again and battery charge level increased.
```
/sys/class/power_supply/ac # echo 1 > charging_enabled
/sys/class/power_supply/ac # cat status
Charging
```

milux, 2017-09-04:
Will be included in next release.

fredsilon, 2017-09-05:
thanks, that was quick :), working great with release 1.0.4

There is only one issue left for my device (maybe it's different with others):

After each reboot, I have to make the control file writable via adb:

`/sys/class/power_supply/ac # chmod u+w charging_enabled`

milux, 2017-09-06:
Otherwise BCL fails to overwrite?
If that is the case, could you please open a new issue for it, because it will require implementing a new option for this.
Thanks for reporting back, will remove the experimental status flag.

closed 2017-09-06 by milux

qtwyeuritoiy, 2017-10-02:
It is an issue for me and I can confirm that it works with the write permission. Currently I've automated the process with Termux but I hope it would be resolven in the future release.