sriharshaarangi/BatteryChargeLimit #47: Reduce necessity to run (most of) the code as root

tim-seoss, 2017-09-12:
It occurred to me that it might be a good idea to split the code into two parts - the existing code could run without root privileges, so long as something (could be a separate helper app - which does need to run as root - but only once per boot, or a shell script, or even a script or other type of permissions tweak which ships with the OS for custom ROMs) changes the necessary permissions on the /sys files to give the app write permissions to the necessary files (permissions only need changing on boot-up, thereafter the main part of the app can run without root privileges).

I've tested this using adb (confirmed that I could stop charging on a jfltexx as a non-root user - having changed the permissions on the necessary files under /sys).

sriharshaarangi, 2017-12-18:
I have tried changing the permissions on a non-rooted stock Oneplus X running Android 6.0 using adb, but it was giving permission denied.
Can you explain the procedure on how you have done it, and the ROM and android version of that device?

milux, 2017-12-20:
@tim-seoss See above
@sriharshaarangi I didn't check it yet, marked it as "todo" because it seems interesting from a theoretical perspective. Some time ago, people on XDA were also talking about "fake rooting", because their control files appeared user-writable, thus root was not necessary for those particular devices.

guillaume-d, 2018-03-11:
Does it mean BCL could run without rooting? (As [also asked on the F-Droid forum](https://forum.f-droid.org/t/use-battery-charge-limit-without-root-setup-possible/1659).)
I could not find any reference to this "fake rooting" on XDA:
Which rights the control files should have for this? I guess
`-rw-rw-r-- system radio`
for  `/sys/class/power_supply/battery/batt_slate_mode` would not be good enough on my Samsung phone, right?

Or does `android.permission.RESET_BATTERY_STATS` suffices on Android < 4.4? (In case it suffers the same fate as its counterpart as outlined in [xda-developers -> Android Development and Hacking -> Android General -> Give us BATTERY_STATS back](https://forum.xda-developers.com/showthread.php?t=2529455)...)

milux, 2018-03-12:
In theory yes.
In your concrete case, the app would either need to be system app (unlikely without root) or the app must somehow be assigned to the user group "radio". Maybe there's a permission for that, I don't know...
Regrading the RESET_BATTERY_STATS: We observed this on a particular phone and were playing around with it, but I don't know of any phone that's really using this a.t.m.

eloyesp, 2019-12-31:
I'm almost sure that it is the way it works the [f-droid privileged extension](https://gitlab.com/fdroid/privileged-extension).

There is a ZIP file that should be loaded with OTA and if it is it available f-droid use that as a bridge to gain superpowers.

Just wanted to add a reference, hoping you find it useful.

The advantage is that I can get rid of the "root notification" in the system tray.

Klimmbimmel, 2022-04-06:
https://forum.xda-developers.com/t/app-root-4-0-battery-charge-limit-v1-1-1.3557002/post-86636735

Maybe adb shell will help you out