sriharshaarangi/BatteryChargeLimit #49: Can this be made so that it tops drawing current?

ro-savage, 2017-09-16:
I'd like to use this app differently to most.

I have DJI Drone and the remote when connected to an android phone will automatically start charging the phone and therefore drawing power from the tiny battery on the controller.

I would like to disable charging completely, so that it no even drawing current from the device (but still transfer data).

Would this be something that could be added to BatteryChargeLimit? (I am using a Samsung S7 Edge)

ps-spark, 2019-08-15:
It would be a great feature for the whole drone + Android community.

milux, 2019-08-19:
The feature is already available for months with the beta 1.2.0 version. If you want to use it, join the telegram beta tester channel. Contact information should be available on XDA.