sriharshaarangi/BatteryChargeLimit #50: Bengali (Bangladesh) Translation!

AdroitAdorKhan, 2017-09-23:
[BatteryChargeLimit.zip](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1326594/BatteryChargeLimit.zip)
Bengali Translated String File is added below! :)

milux, 2017-09-24:
Thanks, will add it asap!
Maybe you can do a pull request next time? That would make things more convenient for everyone.

AdroitAdorKhan, 2017-09-24:
I'm sorry! I'm a noob here in GITHUB! :(

milux, 2018-01-04:
@AdroitAdorKhan I did my very best to include your translation properly.
However, the whitespace in the XML file you provided was terribly screwed up, so there might me error in there... please report back when we release the next version.

milux, 2018-01-05:
@AdroitAdorKhan Can you please check the translation in version 1.1.0?

AdroitAdorKhan, 2018-01-06:
@milux Yes! I checked it, few strings are missing! And, I'm sorry, I used an app to translate on my android device, that might have made whitespace! Can you help me to find the exact string file to translate? :) I'll translate it again properly!

sriharshaarangi, 2018-01-06:
@AdroitAdorKhan [this](https://github.com/sriharshaarangi/BatteryChargeLimit/blob/master/app/src/main/res/values/strings.xml) is the strings file in english, and [this](https://github.com/sriharshaarangi/BatteryChargeLimit/blob/master/app/src/main/res/values-bn-rBD/strings.xml) is the file that contains Bengali translation

AdroitAdorKhan, 2018-01-07:
@sriharshaarangi I'm done changing the English Strings to Bengali! and, pulled a request! :)

sriharshaarangi, 2018-01-08:
@AdroitAdorKhan  Thanks! But I don't see any pull request. Can you please verify and create a pull request again?

AdroitAdorKhan, 2018-01-09:
@sriharshaarangi will you check this time, I haven't figured out all git stuffs yet! Sorry for the inconvenience!

closed 2018-03-13 by milux