sriharshaarangi/BatteryChargeLimit #51: Changing the limit below the charge level while the service is disabled disables charging

sm4rk0, 2017-09-26:
Steps to reproduce:
1. Turn the battery charge limit on, then off.
2. Change the charge limit below the current charge.

Even the in-app status says "DISCHARGING".
App version: 1.0.4
Device: Motorola Moto X (2013)
Control file: pm8921

NetSoerfer, 2017-12-08:
This doesn't appear to happen anymore with 1.05, I just checked.

sriharshaarangi, 2017-12-18:
The batteryReceiver is not unregistering properly, and is receiving the power events even after the foreground service is disabled

sriharshaarangi, 2017-12-29:
@milux @xdnax Please check this commit: [added debug statements for BatteryReceiver](https://github.com/sriharshaarangi/BatteryChargeLimit/commit/7bfa7e931f0f95702308bc42e365cad41059316b)
Now follow the steps given in the OP to reproduce the bug. The batteryReceiver is receiving events even after it is unregistered

xdnax, 2017-12-30:
I too can reproduce this @sriharshaarangi and have been looking for over an hour and cannot fix it. Very weird indeed.

milux, 2018-01-05:
@sm4rk0 Can you please confirm that this bug has disappeared?

sm4rk0, 2018-01-08:
I'm running f-droid build and new version is still not available there. I'll post here as soon as it updates and close the issue if it's fixed.

sriharshaarangi, 2018-05-15:
The issue should be fixed in the update. If you find it otherwise, feel free to re-open this.

closed 2018-05-15 by sriharshaarangi

sm4rk0, 2018-05-15:
Yes, it was fixed. Sorry for forgetting to verify it on time.