sriharshaarangi/BatteryChargeLimit #56: Limit set to 90%, stops charging at 70%

jordanade, 2017-10-05:
Using superman ROM on a Samsung galaxy s7 sm-g930f. Tried both control files available to me (store_mode and batt_slate_mode) and didn't seem to make a difference.

Logs with suggested filter don't include anything at the moment it hits 70. But I can attach complete logs (is there any sensitive information I should edit out?)

jordanade, 2017-10-05:
[image]
[image]
[image]

jordanade, 2017-10-05:
To clarify, the CHARGE_FULL message appears right after I connect the charger, not when it hits 70%

milux, 2017-10-12:
Let me guess... you're using the **store_mode** control file? 🙈 
There are plenty of Samsung devices that show the 70 % behavior when using store_mode... I'm getting tired of all the issues and posts about it - no offence!
Is there another control file that you can select in settings?

jordanade, 2017-10-13:
Yes. I didn't realize I had to restart after changing the control file to batt_slate_mode. That fixed it. You may want to put a message to that effect in the UI.

milux, 2017-10-13:
Wise words... I will consider doing something like that to warn users about the pros/cons of the store_mode control file. Thanks!

closed 2017-10-13 by milux