sriharshaarangi/BatteryChargeLimit #57: Nexus 7 2013 Wi-Fi (razor/flo)

Himura2la, 2017-10-15:
I'm trying to support my device but failed to fing the correct file. Can you please look at my attempts and give some advice?

```sh
1|shell@flo:/sys $ su
root@flo:/sys # find . -iname *charg*
./fs/cgroup/memory/sw/memory.move_charge_at_immigrate
./fs/cgroup/memory/memory.move_charge_at_immigrate
./bus/platform/devices/pm8921-charger
./devices/i2c-0/0-0055/power_supply/battery/charge_full_design
./devices/i2c-0/0-0055/power_supply/battery/charge_avg
./devices/i2c-0/0-0055/power_supply/battery/charge_now
./devices/i2c-0/0-0055/power_supply/battery/charge_full
./devices/i2c-0/0-0055/power_supply/battery/charge_counter
./devices/platform/msm_ssbi.0/pm8921-core/pm8921-charger
./kernel/fast_charge
./kernel/fast_charge/force_fast_charge
./module/msm_otg/parameters/usbhost_charge_mode

root@flo:/sys/bus/platform/devices/pm8921-charger # ls -l
-r--r--r-- root     root         4096 2017-10-15 17:02 modalias
drwxr-xr-x root     root              2017-10-15 16:59 power
lrwxrwxrwx root     root              2017-10-15 17:02 subsystem -> ../../../../../bus/platform
-rw-r--r-- root     root         4096 2017-10-15 16:59 uevent
root@flo:/sys/bus/platform/devices/pm8921-charger #

root@flo:/sys/devices/i2c-0/0-0055/power_supply/battery # ls -l
-r--r--r-- root     root         4096 2017-10-15 16:59 capacity
-r--r--r-- root     root         4096 2017-10-15 17:02 charge_avg
-r--r--r-- root     root         4096 2017-10-15 16:59 charge_counter
-r--r--r-- root     root         4096 2017-10-15 17:02 charge_full
-r--r--r-- root     root         4096 2017-10-15 17:02 charge_full_design
-r--r--r-- root     root         4096 2017-10-15 17:02 charge_now
-r--r--r-- root     root         4096 2017-10-15 16:59 current_now
-r--r--r-- root     root         4096 2017-10-15 17:02 cycle_count
lrwxrwxrwx root     root              2017-10-15 17:02 device -> ../../../0-0055
-r--r--r-- root     root         4096 2017-10-15 17:02 energy_now
-r--r--r-- root     root         4096 2017-10-15 16:59 health
drwxr-xr-x root     root              2017-10-15 16:59 power
-r--r--r-- root     root         4096 2017-10-15 17:02 power_now
-r--r--r-- root     root         4096 2017-10-15 16:59 present
-r--r--r-- root     root         4096 2017-10-15 16:59 status
lrwxrwxrwx root     root              2017-10-15 17:02 subsystem -> ../../../../../class/power_supply
-r--r--r-- root     root         4096 2017-10-15 16:59 technology
-r--r--r-- root     root         4096 2017-10-15 16:59 temp
-r--r--r-- root     root         4096 2017-10-15 16:59 type
-rw-r--r-- root     root         4096 2017-10-15 17:02 uevent
-r--r--r-- root     root         4096 2017-10-15 16:59 voltage_now
root@flo:/sys/devices/i2c-0/0-0055/power_supply/battery # cat status
Charging
root@flo:/sys/devices/i2c-0/0-0055/power_supply/battery # cat charge_full
3626000
root@flo:/sys/devices/i2c-0/0-0055/power_supply/battery # cat charge_full_design
3828000

root@flo:/ # cd /sys/devices/platform/msm_ssbi.0/pm8921-core/pm8921-charger
root@flo:/sys/devices/platform/msm_ssbi.0/pm8921-core/pm8921-charger # ls -l
-r--r--r-- root     root         4096 2017-10-15 17:02 modalias
drwxr-xr-x root     root              2017-10-15 16:59 power
lrwxrwxrwx root     root              2017-10-15 17:02 subsystem -> ../../../../../bus/platform
-rw-r--r-- root     root         4096 2017-10-15 16:59 uevent
```

I doubt that if we write "Not charging" to `status` it will work...

klausmcm, 2017-10-20:
I have this file on my **flo** here which has **0** before and **1** while being plugged in.

`/sys/class/power_supply/ac/online`

I'm on **Paranoid Android 6.0.3**.

Himura2la, 2017-10-20:
@klausmcm thank you for investigation, I'll chech it as soon as I have an access to my tablet. For now, can you please check whether this file has a write permission?

Himura2la, 2017-10-20:
Bad luck. It isn't writable
```
127|root@flo:/sys/class/power_supply/ac # ls -l
lrwxrwxrwx root     root              2017-10-20 20:53 device -> ../../../0-0055
-r--r--r-- root     root         4096 2017-10-20 18:46 online
drwxr-xr-x root     root              2017-10-20 18:46 power
lrwxrwxrwx root     root              2017-10-20 20:53 subsystem -> ../../../../../class/power_supply
-r--r--r-- root     root         4096 2017-10-20 18:46 type
-rw-r--r-- root     root         4096 2017-10-20 18:46 uevent
```

sm4rk0, 2017-10-20:
My Moto X 2013 (ghost) also has the pm8921 and `pm8921_charger` control file (ie. `/sys/module/pm8921_charger/parameters/disabled`) works for me. Did you try it? What's the console output for this:
`ls -l /sys/module/pm*/parameters/`

klausmcm, 2017-10-21:
@Himura2la I have the same issue. However, I also had this issue #46 on my Nexus 5. The app selects the correct control file on my Nexus 5 but the file is also only readable. I did `chmod 644 /sys/class/power_supply/ac/charging_enabled` and the app works fine now. I believe the same would work for the Nexus 7.

@sm4rk0 this is the output for me

```
root@flo:/ # ls -l /sys/module/pm*/parameters/
/sys/module/pm8xxx_regulator/parameters/:
total 0
-rw-------    1 root     root          4096 Oct 20 17:21 debug_mask

/sys/module/pm_8x60/parameters/:
total 0
-rw-rw-r--    1 root     root          4096 Oct 20 17:21 debug_mask
-rw-rw-r--    1 root     root          4096 Oct 20 17:21 sleep_time_override
```

Himura2la, 2017-10-21:
@klausmcm adding a write permission may be a good decision for files that are obviously do switch the charger. `online` looks more like an informative output... But the idea of @sm4rk0 looks great, I'tt check it soon

bayern1, 2018-03-07:
@klausmcm Hi! I have a Nexus 7, and I observed the exact same thing as you mentioned on Oct 19th. However when I try to change the file's content fron 1 to 0, the device just restarts itself, did that happen to you also?

dh29, 2018-04-03:
I have a Nexus 7 **flo** (2013 WiFi). Can't get this app to work, which is a really pity as I wanted to leave the device on permanent wireless charge as I'm using it as a replacement for a Sonos CR100 controller,  which Sonos are about to kill-off.

I'm running lineageos 14.1 (Android 7.1.2).

This is the output for `root@flo:/ # ls -l /sys/module/pm*/parameters/`

```
/sys/module/pm8xxx_regulator/parameters/:
total 0
-rw-------    1 root     root          4096 Oct 20 17:21 debug_mask

/sys/module/pm_8x60/parameters/:
total 0
-rw-rw-r--    1 root     root          4096 Oct 20 17:21 debug_mask
-rw-rw-r--    1 root     root          4096 Oct 20 17:21 sleep_time_override

```
So nothing obvious...
Anybody have any other suggestions? I'm happy to try/evaluate anything.

bayern1, 2018-04-03:
@dh29 I had the exact same problem, there is an XDA thread that you could check to see what you can do to help. Basically they need to know where the 'control file' is in Nexus 7, so that they can do something about it

ihrubab, 2021-01-13:
> @klausmcm Hi! I have a Nexus 7, and I observed the exact same thing as you mentioned on Oct 19th. However when I try to change the file's content fron 1 to 0, the device just restarts itself, did that happen to you also?

I tried to use the /sys/class/power_supply/battery/charge_now and my Nexys 7 rebooted. Did you manage to find a solution to this problem?