sriharshaarangi/BatteryChargeLimit #62: Support for Acer Liquid Z630 (T03)

Mibo5354, 2017-11-06:
Could you please support this phone, what do you need?
Soc: MediaTek MT6735

milux, 2017-11-09:
Please read the OP on XDA carefully, it will inform you about everything you need to know.
Also, check out some (closed) feature requests for devices, and what information has been provided for those devices to become supported. This will help as well. ;)

Mibo5354, 2017-11-10:
It's definitely not the 'online' files listed in #16 
They cause the device to freeze and restart. I will continue to try and find the file.

milux, 2017-11-10:
I will close this meanwhile, report back if you've had any success. You can ask for help in the XDA thread, maybe someone can give you a hint.

closed 2017-11-10 by milux