sriharshaarangi/BatteryChargeLimit #63: App doesn't close after unplugging charger

c64bob, 2017-11-07:
The app doesn't close automatically after the charger is unplugged and if the battery has reached the defined limit, meaning it doesn't charge anymore. When it is still charging and I unplug the charger, the app closes like it is meant to. So somehow the intent "Power Disconnected" isn't received properly when charging is disabled.

Do you need any further information to troubleshoot? Thanks by the way for this great app!

milux, 2017-11-09:
This is a known technical limitation of some control files, not a bug. If the battery level goes below the minimum threshold, it will terminate. Please see the FAQ and XDA discussion for further information!

closed 2017-11-09 by milux

c64bob, 2017-11-13:
Couldn't you use BATTERY_CHANGED intents, to determine whether an external power supply is plugged in more reliably? At least on my phone I can see why ACTION_POWER_DISCONNECTED does not provide a reliable way to determine when to close the app. However, BATTERY_CHANGE does. Of course there is a lot of noise with this intent, but you seem to filter this out already using various mechanisms.

milux, 2017-12-02:
I don't see how BATTERY_CHANGED will help to solve this issue. I think I can follow your idea, but you missed the point: On devices where the charger is disabled **entirely** (those are the ones we are talking about here), there is simply no technical difference between plugged in or not plugged in - at all.
This is a hard technical limitation, there's simply nothing that could circumvent this with one exception: Switching on the charger periodically to observe whether power is still available or not. And this is not a very nice thing to do, plus very error-prone, so I won't implement it. Imho it's much better to wait until the lower limit is reached, and the notification will dismiss itself.