sriharshaarangi/BatteryChargeLimit #69: Automatic Stats Reset when crossing lower threshold

NetSoerfer, 2017-11-27:
On my OnePlus 5 with the latest official LineageOS 14.1, I've set an upper charge limit of 90% and lower limit of 80%, and checked Automatic Stats Reset.
The phone stops charging as it reaches 90% and resets the battery stats correctly.
But when the battery charge dropped below 80% while I was using the phone, I got another stats reset notification even though the phone hadn't been connected to the charger in an hour or so.
That doesn't seem like intended behaviour if I understand it right. Do you need any additional information (logcat etc) or is there no need for that?

milux, 2017-12-02:
Sounds very weird indeed... will try to find out how this can happen. It's definitely a bug, not intended behavior.

NetSoerfer, 2017-12-08:
This doesn't appear to happen anymore with version 1.05. (Is there a changelog for that version?)

milux, 2017-12-09:
I don't know anything about a version 1.0.5? 😕 
Please tell me more. 😆 🙈

milux, 2017-12-09:
Ah, I see, you're talking about a fork of @xdnax, right?
Hope he brings those changes back here, but glad that he found a way to fix this.

NetSoerfer, 2017-12-09:
Oh - his version seems to be on F-Droid under the same name, which is a little confusing. Maybe @xdnax could change the name so both versions aren't mixed when running a store updater?

milux, 2017-12-09:
That can only happen because F-Droid compiles from source...
Our "official" builds are signed with a different developer key, but that doesn't affect F-Droid's builds.

NetSoerfer, 2017-12-09:
Ah, I see. Thanks for explaining!

milux, 2017-12-09:
Ah, well... sorry, I have to correct myself. Actually, the 1.0.5 build **has indeed been released by me**. 🙈 
It was just a minor release adding some more experimental files, and because of the lack of time, I didn't update anything on XDA, just pushed the new version to GitHub. There, the build bot of F-Droid picked it up and built it, therefore you got it...
And maybe, your problem still exists... this minor release didn't introduce any remarkable change of the source code. So, if your problem is **really gone**, than it's probably not because of the update. 😕

xdnax, 2017-12-15:
Just to confirm, the fork I have should not be anywhere like F-Droid. Definitely didn't want to confuse users from the real BCL. 😀

milux, 2017-12-15:
@xdnax Have you ever considered to ask @sriharshaarangi whether he could add you as a contributor to this project?

NetSoerfer, 2017-12-15:
So to get back to the original topic... 😁
The problem persists on my OnePlus 5T with the latest OxygenOS. It doesnt reset the stats when reaching the upper threshold, but it does when crossing below the lower threshold.
I had some issues with stopping charging at the upper threshold on my OnePlus 5 with LineageOS which appear to have masked this issue, but back on LineageOS I can consistently reproduce the automatic stats reset issue.

milux, 2017-12-17:
Ah... I think I got it. On the same device, removing power does **not** close the notification, right?
In that case, that is actually intended behavior... or, let's name it... the best way to handle the limited information.
The reason for this is, that we want to reset only when we are sure that power supply has been removed. However, on many devices this is not possible before trying to refresh the battery charge, which happens when the lower threshold is passed.
I could include an option for people who want their stats to be reset anyway, whenever the upper limit is reached. What do you think?

NetSoerfer, 2017-12-17:
I see. I think an option to reset on hitting the upper limit makes sense:
Battery stats tell me how much battery I've used over time *since the last time I charged the battery*, and how much longer the battery will last with the same average consumption.
While this coincides with unplugging the charger on stock devices, with BCL we separate the two events. I still want to know how I used the battery since I fully charged it, which I did when hitting the upper threshold. Unplugging the charger seems like an arbitrary trigger in comparison.

I'm not complaining, I just don't know the reason why you chose to reset on unplugging, so I'm giving my two cents to provide another perspective.

Another two cents: I think the most understandable way of handling that would be two options: reset on upper limit and reset on lower limit.

Thank you for explaining!

milux, 2017-12-18:
Actually, it's reset on upper limit for certain devices/control files, and reset and lower limit for some other devices/control files. Some disable the charging subsystem altogether - and therefore miss the unplug event - others really just stop charging and can still recognize when the power supply is removed.
Regarding the reason for the design: Some devices cycle multiple times between the limits, especially when they are chosen close to each other. We just didn't want to issue a stats reset every single time the upper limit is hit - That's all.

NetSoerfer, 2017-12-18:
Haha, okay. From a device manufacturer's perspective, I guess if I've gone to lengths to implement a quick charge technology to wear out the battery as fast as possible, I might as well prevent any workarounds for it. Don't want the customer to use the phone any longer than necessary 😈

xdnax, 2017-12-18:
@milux I have reached out a few times on XDA before, but I didn't know where else to message him. I've done some commits updating the icons and stuff and making everything look great, I gave my app a solid redesign to the main page, and a few other backend things I like... If you guys have other features you want added, I can also look at... I am good in Java and not too familiar with Kotlin, so anything major may take me a while. Appreciate the thought though.

sivaraam, 2018-06-16:
Hello. IIUC, I'm also facing the "app notification doesn't go off when charger is unplugged after upper limit is reached" cum "the battery is tricked into charging state even though its unplugged" issues on a Samsung Galaxy J1 ace running Android 5.1.1

milux, 2018-07-11:
@sivaraam Let me quote myself:
"Some disable the charging subsystem altogether - and therefore miss the unplug event - others really just stop charging and can still recognize when the power supply is removed."
Are you 100 percent sure that your device doesn't belong to the former?

sivaraam, 2018-07-13:
Not sure about that. Is there a way to confirm that?

Regardless of that it would be nice to work around this, if possible. Just thinking out loud, would checking for a plugged event when the charge falls below the "Recharge below" threshold fix the issue on those devices which disable the charging system.

sriharshaarangi, 2018-07-14:
@sivaraam That's what the app is doing right? The notification goes away when the lower limit is crossed, and the charger is unplugged(In case the notification doesn't go away immediately on unplug)

sivaraam, 2018-07-14:
> @sivaraam That's what the app is doing right? The notification goes away when the lower limit is crossed, and the charger is unplugged(In case the notification doesn't go away immediately on unplug)

No. It doesn't happen to me, at least. When the charge reaches the lower limit and the app switches the phone back to the charging state *regardless* of whether the charger is connected or not.

sriharshaarangi, 2018-07-17:
Yes, it switches back to charging state. 
If the phone is plugged in it recharges to the limit, otherwise the notification goes away. Isn't the notification going away in case it is not plugged in, and the charge goes below lower limit?

sivaraam, 2018-07-17:
> Isn't the notification going away in case it is not plugged in, and the charge goes below lower limit?

It doesn't. The app switches the battery to charging state when the lower limit is reached and does switch back to discharged state if the device is not plugged in. So, the notification remains there. As a consequence, I experience the weird state in which the battery icon shows the charging symbol but the device is actually discharging.

milux, 2018-10-02:
> It doesn't. The app switches the battery to charging state when the lower limit is reached and does switch back to discharged state if the device is not plugged in. So, the notification remains [...]

Sorry but... I don't get it @sivaraam  😆 
If it goes back to discharged state after it switched the state and recognized that there's no charger, it should terminate. If it doesn't, although it's in discharging state after trying to put the device into charging state, then this is a bug indeed. Is that what you're expecting?
If yes, then we need an unfiltered logcat of that particular timeframe, so we have a chance to find out what's causing this behavior on your device.
If no, then I don't get the problem still... ^^

sivaraam, 2018-10-02:
> Sorry but... I don't get it @sivaraam laughing

You do seem to have got it :wink:

> If it goes back to discharged state after it switched the state and recognized that there's no charger, it should terminate. If it doesn't, although it's in discharging state after trying to put the device into charging state, then this is a bug indeed. Is that what you're expecting?

Yes

> If yes, then we need an unfiltered logcat of that particular timeframe, so we have a chance to find out what's causing this behavior on your device.

What do you mean by unfiltered? Do you need the whole logcat output of other apps too those that aren't related to the app?

BTW, I'm not sure there's a way to get the output via `adb logcat` on my laptop for this case as it needs the phone to discharge while connected to my PC. So, I think I could get the logcat only via an app on my phone. Am I correct? Or am I missing something?

milux, 2018-10-15:
OK, we're making progress... 😆 
By "unfiltered" I mean that you can forget about the filter instructions that we posted on XDA, we need everything from the BCL app.
There are apps for doing that, no need for adb logcat, see XDA OPs.

CongAn, 2022-04-10:
device：Redmi Note5
System Version: Android 9 (MIUI 10)

-------------

When the charging reaches the upper limit, stop charging, but when the power is lower than the lower limit, it does not enter the charging, and the plug-in USB will not enter the charging, so you can only restart the mobile phone.

milux, 2022-04-10:
@CongAn This is a limitation or bug in your ROM, nothing that we can do on our side about this, sorry. :(
You may try other control files, if any.
Also, your issue is not related to this thread, which is stale anyway. Closing.

closed 2022-04-10 by milux

CongAn, 2022-08-20:
> device：Redmi Note5 System Version: Android 9 (MIUI 10)
> 
> When the charging reaches the upper limit, stop charging. However, when the power is lower than the lower limit, it does not enter the charging mode, and it will not enter the charging mode if you unplug and insert the USB again. You can only restart the mobile phone.

@milux I use the configuration of `input_suspend ` in [`control_files.json#L68`](https://github.com/sriharshaarangi/BatteryChargeLimit/blob/ff7360c6048475f1843628c6d47a79313e87f6af/app/src/main/res/raw/control_files.json#L68) to successfully disable and enable charging. I hope to add the remarks of `Redmi Note5` on `Pixel 2/Xiaomi Mi6` of [`control_files.json#L68`](https://github.com/sriharshaarangi/BatteryChargeLimit/blob/ff7360c6048475f1843628c6d47a79313e87f6af/app/src/main/res/raw/control_files.json#L68).