sriharshaarangi/BatteryChargeLimit #70: charged up to threshold -> ping-pong effect

eternalbit, 2017-12-06:
When my device is charged up to the configured threshold (86%) if I leave it plugged in, it will in short intervals (like 20-30s) turn the screen on (like it does, when I plug/unplug it from the charger).
I guess what happens is: it stops charging until the app recognizes that the charge is just below the upper threshold and then starts charging again for the small interval of like 85% to 86% ... this makes it cycle. Whereas it should rather let the charge drop to the specified lower threshold of say 40%.

milux, 2017-12-07:
Which device are we talking about here?
If your lower threshold is somewhat lower than the upper threshold, your device might "resist" against the charger being disabled.
It might switch the charger back on automatically, "fighting" against BCL. If this is really such a device-specific problem, you can try another control file. If that's not an option or doesn't work either, your device might be incompatible.
In case this is an error of BCL, could you please provide log data about what happens when the device is "cycling"?

eternalbit, 2017-12-07:
Device: Samsung Galaxy S4 mini (GT-I9195)

> If your lower threshold is somewhat lower than the upper threshold, your device might "resist" against the charger being disabled.

Did I get it all wrong? Isn't the lower threshold supposed to be 'somewhat lower'?

Didn't try another control file yet. Current is: batt_slate_mode (Samsung). None of the available has a name which suggests it's better suited.

Recorded some log while the "cycling" was going on, but didn't really know what info I was actually looking to log:
12-07 15:31:12.298 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 1000)
12-07 15:31:30.298 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 2000)
12-07 15:31:46.539 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 1000)
12-07 15:32:08.350 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 2000)
12-07 15:32:26.296 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 4000)
12-07 15:32:45.339 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 8000)
12-07 15:33:27.024 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 16000)
12-07 15:34:56.856 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 30000)
12-07 15:36:41.243 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 30000)
12-07 15:38:26.703 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 30000)
12-07 15:40:11.072 D/Charging State(27546): Fixing state w. CHARGE_ON/CHARGE_OFF 96064045 (Delay: 30000)

milux, 2017-12-07:
> Did I get it all wrong? Isn't the lower threshold supposed to be 'somewhat lower'?

You're absolutely right, that's what I meant. :wink:
I only wanted to clarify that if the lower threshold was to near to the upper threshold, there would be little wondering about the observed behavior.

From your logs, it seems that your device is indeed resisting against disabling the charger. Interesting... my "big" Galaxy S 4 is working just fine, but it's on Lineage... I guess you are on stock?

eternalbit, 2017-12-07:
Set the lower threshold to 66%, so that shouldn't be the issue.
I'm still on CyanogenMod 13.0 (Android 6.0.1).

milux, 2017-12-07:
Strange... I would expect this to work fine... 😕 
I don't have any idea why it restores the state. Maybe a newer ROM doesn't have those glitches.
Try to disable the "enforce limit" option. If your device then continues charging until 100 %, that would confirm my guess. You can try out other control files, but if that doesn't help, I can't offer you any better solution, sorry. :(
If available, try `store_mode`, which works just fine for some older Samsung devices. However, on some devices it prevents charging above a certain threshold, typically 70%, until the device is rebooted. Good luck!

milux, 2017-12-09:
I will close this, please let me know if there is anything new about this issue, so I can reopen.

closed 2017-12-09 by milux