sriharshaarangi/BatteryChargeLimit #73: Please suport Xiaomi Mi A1

jochen24, 2017-12-21:
None of the existing control files are working for Xiaomi Mi A1.
I found the following working control file. Please add it to the supported devices.

/sys/devices/soc/qpnp-smbcharger-18/power_supply/battery/battery_charging_enabled

milux, 2017-12-21:
What flags cause the desired change? I assume those:
1 = charging on
0 = charging off
Is that correct?

jochen24, 2017-12-21:
Yes, this is correct.
I tested it via adb shell on my device.
I have written 0 or 1 with echo into the file.
This causes stop charging, but device still recognizes connected to power source.

milux, 2018-01-05:
@jochen24 Can you please check out version 1.1.0 and report if it's working?

jochen24, 2018-01-06:
Hi, I tested it last night and it's working well for my MI A1 with android Oreo. 
Thank you!

closed 2018-01-06 by sriharshaarangi