sriharshaarangi/BatteryChargeLimit #76: Please support Gigaset QV1030

mema0007, 2017-12-24:
"file": "/sys/class/power_supply/battery/status",
"label": "status",
"details": "Gigaset QV1030",
"chargeOn": "Charging",
"chargeOff": "Discharging"

milux, 2018-01-05:
@mema0007 Can you please check out version 1.1.0 and report if it's working?

isaak654, 2018-01-23:
@milux I have a Galaxy S Advance with the same file path "/sys/class/power_supply/battery/status" and with the same chargeOn & chargeOff values, and that profile doesn't work for me, because it keeps charging until 100% instead to stop at 80%.

milux, 2018-01-23:
@isaak654 The existence of a control file is no guarantee that it will work for the particular device.
It **may work**, but two manufacturers or even two different OSes might have completely different implementations.
You should try another control file from the settings - if any - or watch out for some yet unknown control file on your device.

isaak654, 2018-01-23:
@milux My device can only detect the control file related to motorola defy+ in the settings. Why?
Any other details about how to find a working control file for my device?

milux, 2018-01-24:
@isaak654 If this is so, your device is likely not (yet) supported.
The name of the control file ("status") seems to indicate that it is not an active control file, so it's already interesting that this works with the Defy+, but your device is completely different. No wondering here that it doesn't work this time.
Regarding the control file search: I cannot give you much more information as is already given on XDA. **Besides, probing potential control files is a potentially dangerous operation.**
In principle, you open a shell (either in Terminal Emulator or using adb), execute the `su` command to get a root shell, and then play around reading files with promising names (something with charge or battery in the name), reading their contents `cat [some_file]` and rewriting them `echo 1 > [some_file]`.
If your device stops charging at some point - congrats, you found one! :laughing:

mema0007, 2018-01-28:
Sorry for the late Feedback, the Defy was at my parents place of residence. 

It is not working, it does not stop charging when the % is reached. But there is also an third state "Full" in the file: "/sys/class/power_supply/battery/status".

LoadedScissors, 2018-01-29:
I have an elephone p9000, same status file as the defy, charging and not charging are the values used. 
I've tried manually editing the file, including using chmod to set the file from the default read only to writable, but nothing saves. This is true trying with this app, a rooted text editor and bash. I guess there is a file which this one derives its information from. It doesn't appear to be in that folder though.

Any suggestions for other folders to check? 

Data available in that folder is:

Plugged in with full battery (supply status is charging if not full) 
POWER_SUPPLY_NAME=battery
POWER_SUPPLY_STATUS=Full
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CAPACITY=100
POWER_SUPPLY_BATT_VOL=4180
POWER_SUPPLY_BATT_TEMP=340
POWER_SUPPLY_TEMPERATURER=6769
POWER_SUPPLY_TEMPBATTVOLTAGE=616
POWER_SUPPLY_INSTATVOLT=4180
POWER_SUPPLY_BATTERYAVERAGECURRENT=1550
POWER_SUPPLY_BATTERYSENSEVOLTAGE=4180
POWER_SUPPLY_ISENSEVOLTAGE=4187
POWER_SUPPLY_CHARGERVOLTAGE=5184
POWER_SUPPLY_STATUS_SMB=3
POWER_SUPPLY_CAPACITY_SMB=50
POWER_SUPPLY_PRESENT_SMB=0
POWER_SUPPLY_ADJUST_POWER=-1

Unplugged :
POWER_SUPPLY_NAME=battery
POWER_SUPPLY_STATUS=Not charging
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CAPACITY=100
POWER_SUPPLY_BATT_VOL=4178
POWER_SUPPLY_BATT_TEMP=330
POWER_SUPPLY_TEMPERATURER=6712
POWER_SUPPLY_TEMPBATTVOLTAGE=612
POWER_SUPPLY_INSTATVOLT=4178
POWER_SUPPLY_BATTERYAVERAGECURRENT=0
POWER_SUPPLY_BATTERYSENSEVOLTAGE=4178
POWER_SUPPLY_ISENSEVOLTAGE=4176
POWER_SUPPLY_CHARGERVOLTAGE=0
POWER_SUPPLY_STATUS_SMB=3
POWER_SUPPLY_CAPACITY_SMB=50
POWER_SUPPLY_PRESENT_SMB=0
POWER_SUPPLY_ADJUST_POWER=-1

LoadedScissors, 2018-01-29:
I should also note, there is another file that changes with charge state on my phone, located at: /sys/class/power_supply/ac/online

If I change that file to writable using chmod and change it from 1 (charging) to 0 (not charging) with the lead plugged in, charging appears to stop, but it also sadly reboots my phone. Your mileage may vary.

mikesew1320, 2019-04-27:
I have also the same control file location  on my TECNO Camon CX Air with elephoom p9000 @ /sys/class/power_supply/ac/online , I noticed that it becomes "1" while I plugged my phone to a charger and it becomes "0" when i unplugged it. But I didn't tried it with Terminal Emulator for whether or not it disable the cahrge on my phone, I will post here for the result after I tested it.

If you want any thing from me I will give you

REMEMBER that I created a support request on the github for my phone TECNO CX Air and you closed the issue by saying "Test a working control file which is compatible to your phone  and give it to me" so we can continue it here. Isn't it?

mikesew1320, 2019-04-27:
I tried to check whether it stops my phone dlfrone charging or not, I used the command echo "0" > sys/class/power_supply/ac/online vua Terminal Emulator app but ut says me /system/bin/sh: can't create sys/class/power_supply/ac/online: permission dinied error as shown on the screenshot.

Please help me!
[image]

milux, 2019-07-13:
Please note that all "control files" suggested in this issue seem to be status indicators and not "real" control files.
Therefore, unless working control files are observed, this issue is closed.

closed 2019-07-13 by milux

mikesew1320, 2019-07-14:
Sorry for my last request, I found my phone control file at
"/sys/devices/platform/battery_meter/FG_daemon_disable"

I tested it using terminal,magisk modules Acc and IPC also by using your
app as a custom charging control file.

The only issue with this control file is it stack the battery percentage to
the top off value untill a reboot, I think it's very useful if you add
rebootOnUnplug feature for such like charging control files.

Thank you!

milux, 2019-07-16:
@mikesew1320 No more posts about this in that issue please, unless this control file is relevant for Motorola Defy + ;)