sriharshaarangi/BatteryChargeLimit #78: battery is only charged to 70%

voxspox, 2018-01-02:
With the default charge limit 80% my battery only get charged to 70%. App says status discharging then. After disabling the charge limit the device still does not charge. Battery status (system app) confirms this with "not charging".

control file: store_mode (auto selected)
Phone: Samsung Galaxy S7
Content of /sys/.../[battery](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1597433/battery.txt) is maybe helpful.

milux, 2018-01-02:
The hard limit of 70 % on some Samsung devices is a device-specific issue when using `store_mode`, it cannot be fixed in the app. Try to switch to another control file.

closed 2018-01-02 by milux

voxspox, 2018-01-05:
sorry, didn't notice this in the FAQs :(
`batt_slate_mode` seems to work fine with S7! Maybe you want to add this to the auto selection or to the "select control file" menu.