sriharshaarangi/BatteryChargeLimit #81: ChargeLimit crashes during charge

davekra, 2018-01-25:
Hi, I have a OnePlus5 running Oreo 8.0.0 and OOS 5.0.1 and BatteryChargeLimit 1.0.4(20)
BCL control file is set to 'charging_enabled' Generic.  Always write Ctrl. file is enabled.
[logcat.log](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1664969/logcat.log)

BCL sometimes crashes during charging allowing it to charge past the set limit.  In the attached logcat file, charging started at 9:59:41:283
BCL crashed or stopped 10:38:23.389 
power was disconnected 10:39:24.955
power was reconnected 10:39:33.369 
BCL resumed and charging was stopped by BCL at 10:51:32.597

Any ideas on why BCL would stop mid charge then resume after disconnecting/reconnecting power and limit charge like it should?
Thanks,
davidk

davekra, 2018-01-31:
After upgrading to OOS 5.0.2 and reinstalling BCL 1.1.0 I'm still experiencing a crash mid way through the charge.
This time at 13:29:16.440 in the attached logcat
[logcat.log](https://github.com/sriharshaarangi/BatteryChargeLimit/files/1683037/logcat.log)

After unplugging the power and reconnecting, BCL restarted and finished the charge cycle correctly.

Any help would be appreciated.
Thanks,
davidk

milux, 2018-02-01:
Looks like something that shouldn't be null is null sometimes...

davekra, 2018-02-02:
Thank you for re-opening this and looking into it.  Not having used github before I may have closed the thread thinking I was saving the second comment.
I hope you can find a solution.  This app does exactly what I want.  Stays in the background and doesn't need acknowledgment when charging is complete.

milux, 2018-04-29:
Simple Kotlin nullable flaw... Thanks @angelsl for the fix, closing this.

closed 2018-04-29 by milux