sriharshaarangi/BatteryChargeLimit #84: Add Support for AccessGo Phones (mine : AccessGo 4E)

aseptilena, 2018-02-06:
Please add support for the phone.
I have found a possible control file.
It's in /sys/class/power_supply/battery/device/Charger_TopOff_Value.
The file consist of Voltage value(mine: 4110). I guess it has connection with cut off voltage that can representate StateOfCharge

aseptilena, 2018-02-06:
As addition, I have found another file named ADC_Charger_Voltage. The value inside of mine : 4182

aseptilena, 2018-02-06:
The additional file is at the same directory.

aseptilena, 2018-02-06:
I'm so sorry, maybe I found the correct control file. It's in /sys/class/power_supply/ac/. The filename is "online". The value is 1 if my phone is charging.

aseptilena, 2018-02-06:
Oh, my bad, I've just realized that the ADC_Voltage value goes to 4475 if the SoC 100%. Also the "online file in "ac" directory still 1, even the SoC 100%. The "ac" file is only the status of AC plug or Adaptor. So, maybe, the correct file is "Top_Off_Voltage".

aseptilena, 2018-02-06:
But finally I could not find the control file yet, here are the content of uevent file at /sys/class/power_supply/battery/.

aseptilena, 2018-02-06:
POWER_SUPPLY_NAME=battery
POWER_SUPPLY_STATUS=Full
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CAPACITY=100
POWER_SUPPLY_BATT_VOL=4110
POWER_SUPPLY_BATT_TEMP=230
POWER_SUPPLY_TEMPERATURER=7746
POWER_SUPPLY_TEMPBATTVOLTAGE=613
POWER_SUPPLY_INSTATVOLT=4110
POWER_SUPPLY_BATTERYAVERAGECURRENT=860
POWER_SUPPLY_BATTERYSENSEVOLTAGE=4110
POWER_SUPPLY_ISENSEVOLTAGE=4409
POWER_SUPPLY_CHARGERVOLTAGE=4551
POWER_SUPPLY_STATUS_SMB=Not charging
POWER_SUPPLY_CAPACITY_SMB=50
POWER_SUPPLY_PRESENT_SMB=0
POWER_SUPPLY_ADJUST_POWER=-1

aseptilena, 2018-02-06:
The content is shown when the SoC at 100%.

aseptilena, 2018-02-06:
and the charger plugged in

aseptilena, 2018-02-06:
Here is the content of the uevent if SoC at 100% and the charger plugged out.

POWER_SUPPLY_NAME=battery
POWER_SUPPLY_STATUS=Not charging
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CAPACITY=100
POWER_SUPPLY_BATT_VOL=4178
POWER_SUPPLY_BATT_TEMP=230
POWER_SUPPLY_TEMPERATURER=7131
POWER_SUPPLY_TEMPBATTVOLTAGE=580
POWER_SUPPLY_INSTATVOLT=4178
POWER_SUPPLY_BATTERYAVERAGECURRENT=0
POWER_SUPPLY_BATTERYSENSEVOLTAGE=4178
POWER_SUPPLY_ISENSEVOLTAGE=4024
POWER_SUPPLY_CHARGERVOLTAGE=227
POWER_SUPPLY_STATUS_SMB=Not charging
POWER_SUPPLY_CAPACITY_SMB=50
POWER_SUPPLY_PRESENT_SMB=0
POWER_SUPPLY_ADJUST_POWER=-1

aseptilena, 2018-02-06:
And here is the content of the uevent if the SoC at 98% and the charger plugged in.

POWER_SUPPLY_NAME=battery
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CAPACITY=99
POWER_SUPPLY_BATT_VOL=4097
POWER_SUPPLY_BATT_TEMP=230
POWER_SUPPLY_TEMPERATURER=7727
POWER_SUPPLY_TEMPBATTVOLTAGE=612
POWER_SUPPLY_INSTATVOLT=4097
POWER_SUPPLY_BATTERYAVERAGECURRENT=849
POWER_SUPPLY_BATTERYSENSEVOLTAGE=4097
POWER_SUPPLY_ISENSEVOLTAGE=4401
POWER_SUPPLY_CHARGERVOLTAGE=4588
POWER_SUPPLY_STATUS_SMB=Not charging
POWER_SUPPLY_CAPACITY_SMB=50
POWER_SUPPLY_PRESENT_SMB=0
POWER_SUPPLY_ADJUST_POWER=-1

aseptilena, 2018-02-06:
So where is the control file?

aseptilena, 2018-02-16:
Now I see that POWER_SUPPLY_STATUS=Charging
Is the best answer according to the pattern of control_file.json.
So the control file of mine is
"status" in /sys/class/power_supply/battery/status, because there are only two states or possible values(Charging or Not Charging).

milux, 2018-03-06:
Please use the latest stable version of the BCL app, configure your control file as "custom control file" in the settings, and verify correct functionality.
If it works, please report back for integration.

aseptilena, 2018-03-08:
OK, I'll report back after using it.

milux, 2018-03-08:
Thanks, if BCL doesn't start on your device, then try Terminal Emulator and overwrite the file manually with a su shell, please.

aseptilena, 2018-03-10:
I have tried Terminal Emulator, here are my logs.

u0_a95@ACCESSGO_4E:/ $ su
root@ACCESSGO_4E:/ # cat /sys/class/power_supply/battery/status_smb
Not charging
root@ACCESSGO_4E:/ # chmod 774 /sys/class/power_supply/battery/status_smb
root@ACCESSGO_4E:/ # echo "Charging" > /sys/class/power_supply/battery/status_       smb               <
root@ACCESSGO_4E:/ # cat /sys/class/power_supply/battery/status_smb
Not charging

u0_a95@ACCESSGO_4E:/ $ su
root@ACCESSGO_4E:/ # chmod 774 /sys/class/power_supply/battery/status
root@ACCESSGO_4E:/ # cat /sys/class/power_supply/battery/status
Charging
root@ACCESSGO_4E:/ # echo "Not charging" > /sys/class/power_ "Not charging" > /sys/class/power_supply/battery/status
root@ACCESSGO_4E:/ # cat /sys/class/power_supply/battery/status
Charging

u0_a95@ACCESSGO_4E:/ $ su
root@ACCESSGO_4E:/ # chmod 774 /sys/class/power_supply/ac/online
root@ACCESSGO_4E:/ # cat /sys/class/power_supply/ac/online
1
root@ACCESSGO_4E:/ # echo "0" > /sys/class/power_supply/ac/online

aseptilena, 2018-03-10:
The last command, using "online" file, always make my phone restart.

aseptilena, 2018-03-10:
Using the files and parameters above, my test failed with manually control file in BCL.

aseptilena, 2018-03-10:
Even after I have changed the files permission to writeable.

milux, 2018-03-12:
Sorry, but your experiments clearly indicate that none of those files that you've tested will do the job.
You will have to go on searching.
If you find something that actually works, you can post the information here or start a new issue.

closed 2018-03-12 by milux