sriharshaarangi/BatteryChargeLimit #85: Support for Nexus 7

bayern1, 2018-02-12:
Hi! This app seems to serve my purpose very well! However it does not seem to work on my Nexus 7 (2nd gen). My device is rooted, and I installed it from Google play. Could you please help me? Thanks :-)

milux, 2018-03-06:
Closed, please see XDA OP for instructions.

closed 2018-03-06 by milux

bayern1, 2018-03-07:
Hi! I checked the thread. It seems that the only control file the app can find is {
    "file": "/sys/class/power_supply/battery/status",
    "label": "status",
    "details": "Motorola Defy+ (MB526)",
    "chargeOn": "Charging",
    "chargeOff": "Discharging",
    "experimental": true
  },
What happens is that the file is readable only, and even if I changed it to writable, I can not really modify the content, it auto updates itself to be either 'Charging' or 'Not charging'.

However, I did find that in the file '/sys/class/power_supply/ac/online', the content is '1' whenever the charger is plugged, and '0' when it is not. However, when I change the file writable and modify the content, the device just restarts...

Any clue on what I can do next?

milux, 2018-03-08:
It is very likely that neither one of these files is going to work.
Their very names suggest this.
All you can do is to search harder or wait for someone else to find the right file - maybe. Sorry.

bayern1, 2018-03-09:
Ah, I see... Thanks anyways!