sriharshaarangi/BatteryChargeLimit #86: Nexus 5 on latest LineageOS (14.1) is ignoring the charge limit

Reed97123, 2018-02-17:
I set the limit to 40%, but it kept charging after it hit 40%.

Any suggestions on what I can try or how I can debug the issue?

Reed97123, 2018-02-17:
I used adb to shell into the phone and found the file referenced by the json strings for the phone.  I wanted to attempt to change the value of the changing_enabled file manually to see if it worked (using echo 0 > charging_enabled).

The file permissions were 444 (read-only) so I set it to 644 so I could edit it.  I confirmed that using echo I could turn it on and off.  Then after I did that your app started working.  

Was it as simple as the app not checking for appropriate file permissions before attempting to edit the file?

Reed97123, 2018-02-25:
This continues to be a problem as the file gets reverted back to 444 periodically, perhaps upon boot or other events.  It works again if I go in and shell in with adb and execute "chmod 644 charging_enabled" as root.

It looks like this is the command that changes the state:
val switchCommands = arrayOf("mount -o rw,remount $file", "echo \"$newState\" > $file")

Would this work to change the permissions of the file before the write?
val switchCommands = arrayOf("mount -o rw,remount $file", "chmod 644 $file", "echo \"$newState\" > $file")

I could test this out if someone could instruct me on how to compile the code.

Reed97123, 2018-02-27:
I confirmed the change works.  I created a fork and pull request.

milux, 2018-03-06:
I will have a look at your PR as soon as possible. Thanks for your contribution!

closed 2018-03-06 by milux