sriharshaarangi/BatteryChargeLimit #87: App does not stop charging SM-G920A

TeracottaShishi, 2018-02-19:
I am using an SM-G920A with a SM-G935F battery (3600 mAh versus 2550 mAah). The battery stats have been recalibrated but the charge is still kernel locked because G920A has a locked bootloader.

I have observed when the phone reaches 80%, the upper limit to initiate charging, there is a fierce charging enabled/charging disabled cycle. Here is what appears under the logcat output:

D/BatteryService( 4484): !@BatteryListener : batteryPropertiesChanged!
D/BatteryService( 4484): level:97, scale:100, status:3, health:2, present:true, voltage: 4303, temperature: 271, technology: Li-ion, AC powered:false, USB powered:false, POGO powered:false, Wireless powered:false, icon:17303376, invalid charger:0
D/BatteryService( 4484): online:1, current avg:267, charge type:0, power sharing:false, high voltage charger:false, capacity:280000, current_now:177
D/BatteryService( 4484): Sending ACTION_BATTERY_CHANGED.
I/MotionRecognitionService( 4484): On Battery, 0
I/MotionRecognitionService( 4484): isTAConnected | current backoffstate  = 0
D/MotionRecognitionService( 4484): TA disconnected,  0
I/MotionRecognitionService( 4484): skip setTransmitPower. 
D/MotionRecognitionService( 4484):   cableConnection= 0
D/KeyguardUpdateMonitor( 4735): received broadcast android.intent.action.BATTERY_CHANGED
I/PERF    ( 4735): received broadcast android.intent.action.BATTERY_CHANGED
D/KeyguardUpdateMonitor( 4735): handleBatteryUpdate
D/BatteryMonitor( 4704): new battery level: 97
V/EmergencyMode( 5096): [EmergencyStateReceiver] onReceive : android.intent.action.BATTERY_CHANGED
V/EmergencyMode( 5096): [EmergencyStateReceiver] ACTION_BATTERY_CHANGED mBatteryLevelForLogging = 97
D/QSTile.TorchLightTile( 4735): ACTION_BATTERY_CHANGED - Level :: 97, Status :: 3
D/RcsSettingsUtils( 4704): new battery level: 97
D/CommonServiceConfiguration( 4704): getStringValueSetting
D/BatteryMeterView( 4735): ACTION_BATTERY_CHANGED : level:97 status:3 health:2
D/RcsSettingsUtils( 4704): read setting battery level: NONE
D/BatteryMeterView( 4735): ACTION_BATTERY_CHANGED : level:97 status:3 health:2
D/BatteryMeterView( 4735): ACTION_BATTERY_CHANGED : level:97 status:3 health:2
D/ApplicationPolicy( 4484): isStatusBarNotificationAllowedAsUser: packageName = com.slash.batterychargelimit,userId = 0
V/ApplicationPolicy( 4484): isApplicationStateBlocked userId 0 pkgname com.slash.batterychargelimit
W/NotificationService( 4484): Pray mode not found android.content.pm.PackageManager$NameNotFoundException: Application package com.sec.android.settings.praymodewidget not found
D/BatteryStatsImpl( 4484): poolsize= 1, curBatteryRealtime= 1217277000, mUpdateTime= 1206820000, mTimeout= 0, mNesting= 1, mTotalTime= 218740000

milux, 2018-03-06:
I can't see any BCL output in this log. Sorry, but imho it lacks any relevant information to understand/fix a potential issue.
Apart from that, how should BCL work when your BL is locked? (Which means that there is also no root access, right?)

TeracottaShishi, 2018-03-09:
I have full (?) root access, I'm using a combination image set with
selinux permissive and a pushed supersu. The only exception seems to
be unsigned system apps, which the locked bootloader won't allow even
once selinux is set to permissive.

What I notice is the /sys/class_power_supply/battery folder doesn't
seem to be writable.
I know this because I cannot manually edit the battery_capacity_max
bin file, to account for the 3600 mAh Galaxy S7 edge battery I have
installed into the device.

If I attempt to modify the kernel in some way, such as
slowing/speeding up the CPU, the phone shuts down and won't boot. If I
try to charge it, it won't recognize the GS7e battery, I have to
install a 2550 mAh battery and Odin everything to correct the
attempted kernel change.

At this point, the battery stats have readjusted back to 2550 mAh,
even though the modified batterystats.bin normally survives
reflashing. That makes me think the real battery parameters are hard
coded into the /sys/ folder. So, I wonder if the /sys/ folder under
the locked bootloader goes beyond normal su privileges.

Your app works perfectly on the unlocked G935F. But, I can't tell if
this is a Qualcomm issue, or a locked bootloader issue.

JP

milux, 2018-03-09:
If you have root access, then you should be able to to anything on your system.
AFAIK, the limitations you describe are not caused by the locked BL, but maybe I'm mistaken altogether.
However, as this is a super-hacky super-specific problem, it is definitely out of scope for this project.
I hope you get it sorted out, happy hacking!

closed 2018-03-09 by milux