sriharshaarangi/BatteryChargeLimit #88: Would it be possible to track battery aging by having a detailed log of the charges performed?

CatalinCaranfil, 2018-02-20:
My feeling is that on most devices the same percentage (let's say 80%) will map to a voltage that will change in time mapping the aging trend. Even if at this time the feature will not be used it would make sense to keep a log with all charges (and maybe later interpret it with this app or a separate one).

So bottom line - can we have a log/database at least with the readings at the top of the charge? (something like DATETIME: CHARGING 80% 4.182V / NOT CHARGING  80% 4.175 and so on) And in the first stage mage a very simple way to share/send that file out of the app to analyze it?

milux, 2018-03-06:
Good idea, maybe future work.

digitalcircuit, 2018-10-27:
For design consideration, there's a few apps that do something simple like this, [e.g. Battery Log](https://play.google.com/store/apps/details?id=kr.hwangti.batterylog ) (*simple: ruling out Better Battery Stats, which is logging everything*).

Unfortunately, it'd require a persistent service to log every percentage change, but it doesn't seem to have too much impact.  Regardless, the UI and/or format might be useful for reference should anyone try tackling this.