sriharshaarangi/BatteryChargeLimit #89: Feature request: Support qcom-battery/restricted_charging alternative

80tf3262p8, 2018-02-22:
Phone: OnePlus 5 A5000
ROM: LineageOS 15.1 UNOFFICIAL 20180217 (may affect others as well)
Baseband from Oreo ROM

The built-in methods don't work for me, even though the phone appears to stop charging. (In reality, it's still getting a charge and I can see the voltage increasing.)

I found /sys/class/qcom-battery which has two control files:
1) restricted_current: defaults to 1000000
2) restricted_charging: enables (1) or disables (0) the specified current limit from restricted_current

The app could set restricted_current to 0 and then toggle restricted_charging off and on.

For now, I've set a startup script to change restricted_current to 0 and customized the app to change /sys/class/qcom-battery/restricted_charging with Enable Value: 0, Disable Value: 1, and I have all the other checkboxes disabled. Not needing the startup script would make this more accessible for others.

The phone will continue to report that it is charging even when charging is disabled through this method.

80tf3262p8, 2018-02-22:
Alternatively, restricted_charging could remain enabled and restricted_current could change. This would allow the current to be limited at all times, which is good for those who are worried about fast charging wearing down their battery faster than necessary. (I think I'm going to make my startup script, in the meantime, just turn on restricted_charging instead, so I can have this happen.)

milux, 2018-03-06:
Interesting. This will require changed semantics of BCL, i.e. a significant amount of implementation effort.
Could also lead to general support of charging current control for supported devices.
I regard it future work.

zhongfu, 2018-11-13:
Can confirm that on the OnePlus 6, this seems to prevent battery charging while still drawing power for phone power requirements (similar to `battery_charge_limit` on my old Nexus 6P), but only on standard chargers (i.e. not with Dash Charge).

With `restricted_current` set to 0 and `restricted_charging` set to 1, AccuBattery reports an average current draw of 20mA for charging the battery while idle, and around -85mA while running AnTuTu benchmark.

Similarly, my USB current meter reports anywhere from 0.06A draw from the port (airplane mode) to 1.37A (while benchmarking) on a 2.4A charger (although the OP6 only seems to do BC1.1/1.2, i.e. limited to 1.5A).

This would be a pretty great alternative to using `input_suspend` or `charging_enabled`, since you'll be able to maintain the charge instead of having it drop until the lower charge limit, then get charged up to the upper charge limit. Especially helpful when your phone is churning away at something overnight, since you won't wreck your battery by charging that last few % over and over again.

jakubiakdev, 2021-01-31:
This seems to not be OnePlus exculsive feature. I just checked on my Redmi Note 7 and files were there