sriharshaarangi/BatteryChargeLimit #90: Purify incompability

RoboBlawck, 2018-02-26:
So, if any of you are trying to use the battery charge limit in conjunction with the purify app, please disable purify as it will freeze the battery charge service in an attempt to "boost battery".

Also, a good way of seeing if it's the service that's being disabled and not just your control file being wonky, simply manually test the control file states using your preferred root explorer. If the control file works while being charged, but the app doesn't, it's probably something about the service not functioning properly. In my case, I found out that Purify was the culprit in disabling my service, and so I disabled it right away.

milux, 2018-03-06:
Thanks for the information. What do you expect us to do with it?
I think we should add this information to the Q/A. Do you see a necessity for implementation of any countermeasures?

RoboBlawck, 2018-03-10:
I think the best countermeasure would be to somehow inform the user specifically that the service is being disabled from within the app. While adding the information to the Q/A is good, it's not as necessarily effective at helping users identify the root of the issue. From my experience with Purify it disabled the service without me even knowing, and it only took around a day of testing to find the cause of it. However, for less tech-savvy users it perhaps would take longer since they're seeing the result of the problem(I.E. the phone charging over the limit) but not exactly how it got there.

sriharshaarangi, 2018-05-15:
I think Purify is an (apps-killing) app like Greenify, and obviously you need to whitelist BCL from being killed by the app.  There is nothing we can do from stopping such apps killing our app. 
If you feel otherwise and find something, open the issue and we will sort it out.

closed 2018-05-15 by sriharshaarangi