sriharshaarangi/BatteryChargeLimit #91: Please support Samsung Galaxy Ace 2

wudu80, 2018-02-27:
Samsung Galaxy Ace 2 (GT-I8160), Android 6.0.1 Lineage OS 13.0

"file": "/sys/devices/platform/ab8500-i2c.0/ab8500-charger.0/power_supply/ab8500_usb/status",
"label": "status",
"details": "GT-I8160",
"chargeOn": "Charging",
"chargeOff": "Discharging"

milux, 2018-03-06:
Have you tested correct functionality via the "custom control file" configuration in the settings?
I ask because files called "status" or the like often turn out to be just information providers without any active control functionality, so I want this to be properly tested before we publish this.

wudu80, 2018-03-06:
If I start the app, then the following message appears "Your device isn't supported (yet)!" and the app closes immediately.Therefore I can't check out the functionality of the control file.

milux, 2018-03-08:
Oh, ehm... that might actually be possible. ^^
Have you verified that changing the suggested file has the desired effect via Terminal Emulator?
Try to issue the command `echo Discharging > /sys/devices/platform/ab8500-i2c.0/ab8500-charger.0/power_supply/ab8500_usb/status` on a root shell (issue the `su` command before).