sriharshaarangi/BatteryChargeLimit #95: Crash on usb connection

sozonnyk, 2018-03-10:
Android 4.0.4, SDK 15, Sony Xperia Ray, ST18i
I'm experiencing service crash on usb charger connection, with the following error.

```E/AndroidRuntime( 2294): FATAL EXCEPTION: main
E/AndroidRuntime( 2294): android.content.ReceiverCallNotAllowedException: IntentReceiver components are not allowed to regi
ster to receive intents
E/AndroidRuntime( 2294):        at android.app.ReceiverRestrictedContext.registerReceiver(ContextImpl.java:116)
E/AndroidRuntime( 2294):        at android.app.ReceiverRestrictedContext.registerReceiver(ContextImpl.java:110)
E/AndroidRuntime( 2294):        at com.slash.batterychargelimit.SharedMethods.isPhonePluggedIn(SharedMethods.kt:141)
E/AndroidRuntime( 2294):        at com.slash.batterychargelimit.SharedMethods$startService$1.run(SharedMethods.kt:230)
E/AndroidRuntime( 2294):        at android.os.Handler.handleCallback(Handler.java:605)
E/AndroidRuntime( 2294):        at android.os.Handler.dispatchMessage(Handler.java:92)
E/AndroidRuntime( 2294):        at android.os.Looper.loop(Looper.java:137)
E/AndroidRuntime( 2294):        at android.app.ActivityThread.main(ActivityThread.java:4441)
E/AndroidRuntime( 2294):        at java.lang.reflect.Method.invokeNative(Native Method)
E/AndroidRuntime( 2294):        at java.lang.reflect.Method.invoke(Method.java:511)
E/AndroidRuntime( 2294):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:784)
E/AndroidRuntime( 2294):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:551)
E/AndroidRuntime( 2294):        at dalvik.system.NativeStart.main(Native Method)```

```Packages:
  Package [com.slash.batterychargelimit] (2beda410):
    userId=10071 gids=[]
    sharedUser=null
    pkg=Package{2bf871d8 com.slash.batterychargelimit}
    codePath=/data/app/com.slash.batterychargelimit-1.apk
    resourcePath=/data/app/com.slash.batterychargelimit-1.apk
    nativeLibraryPath=/data/data/com.slash.batterychargelimit/lib
    versionCode=22
    versionName=1.1.0
    dataDir=/data/data/com.slash.batterychargelimit
    targetSdk=25
    supportsScreens=[small, medium, large, xlarge, resizeable, anyDensity]
    timeStamp=2018-03-10 13:57:50
    firstInstallTime=2018-03-10 13:57:51
    lastUpdateTime=2018-03-10 13:57:51
    installerPackageName=cm.aptoide.lite
    signatures=PackageSignatures{2beda4e8 [2bedce70]}
    permissionsFixed=true haveGids=true
    pkgFlags=0x0 installStatus=1 stopped=false enabled=0
    grantedPermissions:
      android.permission.BATTERY_STATS
      android.permission.RECEIVE_BOOT_COMPLETED
```

franzalex, 2018-08-20:
I use the same phone and was experiencing the same problem so I created an Automate flow to restart the app after the charger is connected. Though created with the Sony Ericsson Xperia Ray in mind, it can be used for any device where BCL crashes upon connection of the power cable.

**Installation Steps**

1. Install [**Automate**](https://play.google.com/store/apps/details?id=com.llamalab.automate).
2. Download and install the [flow I created](https://llamalab.com/automate/community/flows/23748). [[direct link]](https://llamalab.com/automate/community/api/v1/flows/23748/data/Restart+Battery+Charge+Limit.flo)
3. Run the flow.
4. Grant the necessary permissions.
5. Optional tuning:  
    - The delay block (block no. 3) can be adjusted to change the duration to wait after power is connected.  
I set it to 15 seconds to give enough time for Battery Charge Limit to crash before restarting it.
    - Remember to set your preferred maximum charge in the App Start block (block no. 5).  
    Change the **92** in the extras field to your preferred maximum charge limit. I.e. ```{"android.intent.extra.TEXT": <your value here>}```

**Note to Developers:** I used the start activity method because there's no running app to receive the intent after the application has crashed.

sriharshaarangi, 2018-08-25:
This might fix the issue: https://github.com/sriharshaarangi/BatteryChargeLimit/commit/c9735b4ebf8cc9408770dfc1ef7f663ccd7960d9
Can you try this [apk](https://drive.google.com/file/d/1wFctY1w624mk8Zv2YjPgP5CehgjTqFbR/view?usp=sharing) and report back if it working? (You need to uninstall the existing version first)

franzalex, 2018-08-27:
Hello @sriharshaarangi
I have just tested the linked APK on my Xperia Ray and its working fine so far.
I'll keep it running for a couple of days and report any issues if I find any.

sozonnyk, 2018-08-27:
I can confirm it is working with my phone as well. Thanks @sriharshaarangi

closed 2018-10-02 by milux