sriharshaarangi/BatteryChargeLimit #96: Essential Phone PH-1 (mata) file info

TiloGit, 2018-03-10:
hi, 

try to check on my PH-1. here what I found while charging with my computer USB.  Let me know how I can pin down the file. Should be similar Pixel 2 i think. 

```console
mata:/sys $ find . -iname *charg*
find: No ./fs/cgroup: Permission denied
find: No ./fs/pstore: Permission denied
./bus/platform/drivers/qcom,smb138x-charger
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/main/constant_charge_current_max
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/battery/step_charging_step
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/battery/charger_temp_max
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/battery/charger_temp
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/battery/charge_done
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/battery/charge_type
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/battery/constant_charge_current_max
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qcom,qpnp-smb2/power_supply/battery/step_charging_enabled
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qpnp,fg/power_supply/bms/constant_charge_voltage
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qpnp,fg/power_supply/bms/charge_now_raw
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qpnp,fg/power_supply/bms/charge_full_design
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qpnp,fg/power_supply/bms/charge_now
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qpnp,fg/power_supply/bms/charge_full
./devices/soc/800f000.qcom,spmi/spmi-0/spmi0-02/800f000.qcom,spmi:qcom,pmi8998@2:qpnp,fg/power_supply/bms/charge_counter
./devices/soc/c1b5000.i2c/i2c-7/7-0008/c1b5000.i2c:qcom,smb138x@8:qcom,smb138x-parallel-slave@1000/power_supply/parallel/charger_temp_max
./devices/soc/c1b5000.i2c/i2c-7/7-0008/c1b5000.i2c:qcom,smb138x@8:qcom,smb138x-parallel-slave@1000/power_supply/parallel/charger_temp
./devices/soc/c1b5000.i2c/i2c-7/7-0008/c1b5000.i2c:qcom,smb138x@8:qcom,smb138x-parallel-slave@1000/power_supply/parallel/charge_type
./devices/soc/c1b5000.i2c/i2c-7/7-0008/c1b5000.i2c:qcom,smb138x@8:qcom,smb138x-parallel-slave@1000/power_supply/parallel/constant_charge_current_max
./devices/soc/c1b5000.i2c/i2c-7/7-0008/c1b5000.i2c:qcom,smb138x@8:qcom,smb138x-parallel-slave@1000/power_supply/parallel/charging_enabled
find: No ./devices/system/cpu/cpu0/rq-stats: Permission denied
find: No ./devices/virtual/misc/hw_random: Permission denied
find: No ./devices/virtual/block/zram0: Permission denied
find: ./devices/virtual/graphics/fb0/lineptr_value: Permission denied
find: ./devices/virtual/graphics/fb0/msm_fb_persist_mode: Permission denied
find: ./devices/virtual/graphics/fb0/idle_time: Permission denied
find: ./devices/virtual/graphics/fb1/hpd: Permission denied
find: ./devices/virtual/graphics/fb1/hdcp/tp: Permission denied
find: ./devices/virtual/graphics/fb1/lineptr_value: Permission denied
find: ./devices/virtual/graphics/fb1/res_info: Permission denied
find: ./devices/virtual/graphics/fb1/s3d_mode: Permission denied
find: ./devices/virtual/graphics/fb1/product_description: Permission denied
find: ./devices/virtual/graphics/fb1/dynamic_fps: Permission denied
find: ./devices/virtual/graphics/fb1/msm_fb_persist_mode: Permission denied
find: ./devices/virtual/graphics/fb1/idle_time: Permission denied
find: ./devices/virtual/graphics/fb1/vendor_name: Permission denied
find: ./devices/virtual/graphics/fb2/lineptr_value: Permission denied
find: ./devices/virtual/graphics/fb2/msm_fb_persist_mode: Permission denied
find: ./devices/virtual/graphics/fb2/idle_time: Permission denied
find: ./devices/virtual/graphics/fb3/hpd: Permission denied
find: ./devices/virtual/graphics/fb3/lineptr_value: Permission denied
find: ./devices/virtual/graphics/fb3/res_info: Permission denied
find: ./devices/virtual/graphics/fb3/s3d_mode: Permission denied
find: ./devices/virtual/graphics/fb3/msm_fb_persist_mode: Permission denied
find: ./class/sidecar/power_control: Permission denied
./class/qcom-battery/restricted_charging
find: ./power/wake_unlock: Permission denied
find: ./power/wake_lock: Permission denied
./firmware/devicetree/base/soc/qcom,spmi@800f000/qcom,pmi8998@2/qpnp,fg/qcom,fg-recharge-soc-thr
./firmware/devicetree/base/soc/qcom,spmi@800f000/qcom,pmi8998@2/qpnp,fg/qcom,fg-auto-recharge-soc
./firmware/devicetree/base/soc/qcom,spmi@800f000/qcom,pmi8998@2/qcom,qpnp-smb2/qcom,auto-recharge-soc
./firmware/devicetree/base/soc/qcom,spmi@800f000/qcom,pmi8998@3/qpnp-labibb-regulator/qcom,ibb@dc00/qcom,qpnp-ibb-en-discharge
./firmware/devicetree/base/soc/qcom,spmi@800f000/qcom,pmi8998@3/qpnp-labibb-regulator/qcom,lab@de00/qcom,qpnp-lab-max-precharge-time
find: ./kernel/slab/L2TP/IPv6: No such file or directory
find: ./kernel/slab/L2TP/IP: No such file or directory
find: No ./kernel/debug: Permission denied
find: ./kernel/dload/emmc_dload: Permission denied
find: ./kernel/uevent_helper: Permission denied
./module/smb138x_charger
find: ./module/lowmemorykiller/parameters/adj: Permission denied
find: ./module/lowmemorykiller/parameters/cost: Permission denied
find: ./module/lowmemorykiller/parameters/vmpressure_file_min: Permission denied
[... 47 lines cut ...]
-rw-r--r-- 1 root   root   4.0K 2018-03-10 09:08 system_temp_level
-r--r--r-- 1 root   root   4.0K 2018-03-10 09:05 technology
-r--r--r-- 1 root   root   4.0K 2018-03-10 09:05 temp
-r--r--r-- 1 root   root   4.0K 2018-03-10 09:05 type
-rw-r--r-- 1 root   root   4.0K 2018-03-10 09:15 uevent
-r--r--r-- 1 root   root   4.0K 2018-03-10 09:15 voltage_max
-r--r--r-- 1 root   root   4.0K 2018-03-10 09:05 voltage_now
-r--r--r-- 1 root   root   4.0K 2018-03-10 09:15 voltage_qnovo

1|mata:/sys $ cat /sys/class/power_supply/battery/input_suspend                                                                                                                
0

1|mata:/sys $ cat /sys/class/power_supply/battery/charge_done                                                                                                                  
0

1|mata:/sys/class/power_supply/battery $ cat current_now                                                                                                                              
-472167

mata:/sys/class/power_supply/battery $ cat voltage_now                                                                                                                                
4175038

mata:/sys/class/power_supply/battery $ cat capacity                                                                                                                                   
79

```

milux, 2018-03-19:
`/sys/class/power_supply/battery/input_suspend` (Pixel 2 and others) is already supported by BCL.
Did you try out if this already works out of the box before opening this issue?

TiloGit, 2018-03-19:
Hi, so far seem working fine. 
But couldn't fine Essential in the JSON phone list: https://github.com/sriharshaarangi/BatteryChargeLimit/blob/master/app/src/main/res/raw/control_files.json

I opened this issue to aid the official support by BCL. Sorry if that was the wrong approach.

milux, 2018-03-19:
Good, then your phone is "officially supported". 😉 
We cannot list each and every supported smartphone. The labels are just meant as some sort of a hint.
Thanks for your information, I will close this then.

closed 2018-03-19 by milux