sriharshaarangi/BatteryChargeLimit #100: Disabling from notification not synced in app

sm4rk0, 2018-03-21:
Maybe this is a feature, not a bug. In that case, a toast should say "Battery Limit Service Temporarily Disabled!".

Steps to reproduce:
1. Connect the device to a charger
2. Ensure the Battery Charge limit service is enabled
3. Disable it in notification

Expected behavior:
* Battery Charge Limit switch changes to "off" position in application.
* After reconnecting the charger, Battery Charge Limit should stay disabled

Actual behavior:
* Battery Charge limit switch stays in "on" position in application.
* After reconnecting the charger, Battery Charge Limit is enabled again

milux, 2018-03-21:
I've noticed this inconsistent behavior as well.
Will fix this with the next release.

sriharshaarangi, 2018-03-27:
That was added as a feature by @milux. After the upper limit is reached, the phone starts discharging, and we can remove the notification before waiting for the phone to reach the lower limit.

We can add the expected behaviour in OP to disable BCL if the phone hasn't reached the limit yet, and continue the current implementation if the phone is discharging after the limit is reached. In this case, we can change 'Disable' to 'Dismiss'

milux, 2018-03-28:
@sriharshaarangi I don't think that I implemented this. I think it was an idea of @xdnax, but maybe I've just forgotten why and when I did it. :laughing: 
Sounds good, I will do it in the way you suggested. :+1:

sriharshaarangi, 2018-05-17:
When the 'Disable' button is hit
1. Before the limit is reached: It allows the phone to temporarily charge to 100%(or until unplugged), and again enables on replugging.
2. After the limit is reached

- If the phone is still plugged in, the phone again starts charging until the limit.
- If the phone is unplugged, the notification gets dismissed.

In the first case, we can name it 'Disable temporarily'
In second case, we will not be able to detect whether it is unplugged or plugging. Naming it 'Dismiss' won't have a negative effect on the other case.

@milux  I have implemented this here 6c411f7e4fa0b095963a340cb258b6bab2cdfd56. Do you have any better naming convention?

sm4rk0, 2018-05-17:
How about this UX idea:
* add "Swipe notification to dismiss" for temporarily disabling the service with a text like "Swipe to charge once to 100%" instead "Battery will stop charging once limit... Please wait..." which is BTW too long for my device to show (Motorola Moto X)
* leave the `Disable` button to disable the service until user enables it again
* the notification appears whenever the USB cable is plugged, so the user can enable the service if they wish

sriharshaarangi, 2018-05-22:
- That notification cannot be swiped, as a foreground service is attached to it(to check for battery percentage)
- The app shouldn't be running if the user disabled the service in the app
- There is a widget to disable/enable service in 1-click

sm4rk0, 2018-05-22:
Maybe I didn't express myself clearly, but here are my points/arguments:
* There's no need for the foreground service if the user disables it (by swiping the notification).
* The `Disable` button would be there while the app (and service) is running, for disabling it permanently, not while the app is not running.
* Widget might be helpful, but why not putting all control into the notification when it already has to be there (for the foreground service)?

532910, 2020-11-17:
I believe this is the feature and I exepect it will be enable after power cable re-plug. Toast should be "Battery Limit Service Temporarily Disabled!"